carla-simulator/carla #6761: Carla Vehicle Skeleton Related Question

stevenbrills, 2023-09-08:
I'm interested in creating a specific type of vehicle skeleton to test some autonomy capability. The base skeleton of this vehicle does not match the four-wheeled setup expected/provided by Carla since other elements of the vehicle can be actuated.

How does Carla interface with the skeleton? Does Carla only support the four-wheeled skeleton that it provides and expect the user to simply bind meshes to the default skeleton so that references to names/calls to the skeleton are maintained?

If support for a different skeleton would have to be added, how should I go about it? I'm looking for some high level information on what components on Carla's side would need to be changed to support a different skeleton in the cleanest manner possible and how they interact with the 3D asset.

gsmario, 2024-05-24:
Hello! IT depends on how many wheels your vehicle have. As long as you don't modify the 5 important bones (vehicle base, and each wheel) you can add any other bone you like, to attack doors or somethingl ike that. 
We also have a 6 wheel vehicle, a truck, but I'm not sure if thats where you want to go, or you're looking for something else.
If you show me the skeleton I probably can help you a bit more.

Thanks in advance.

gsmario, 2024-05-31:
I'm closing this issue, feel free to open it again or contact me via discord.
Sorry for the late reply, and thanks for your time.

closed 2024-05-31 by gsmario