carla-simulator/carla #6763: Parking Lot Navigation

ElSahn, 2023-09-09:
Hello, in the following **scenario** I would like the **ego-vehicle** to **maneuver** the **static vehicle** by going to the left lane.
[image]

Normally such maneuver is done by getting a waypoint on the left lane. However, I found out that there are no explicit lanes/waypoints inside the parking lot (as seen in the below figure).
[image]
Are there any suggestions?
What came up into my mind is to add lanes/waypoints inside the parking lot (which requires building CARLA from source). I am not sure how difficult that should be tho.
Thanks in advance.

Dragonking3x, 2023-10-03:
If you don't need the control the car with python, you can use [scenariorunner](https://github.com/carla-simulator/scenario_runner)
You can run an openSCENARIO with scenariorunner, and set waypoints with XYZ, road related, or lane related.
For your example, road related with a shifted t would do the trick.

ilkeery, 2024-07-19:
I am dealing with a similar problem, could you solve the problem? Is there any way to use Controller in the parking lot?