carla-simulator/carla #6772: How to generate dataset labels with the same format as nuscenes

WangzcBruce, 2023-09-13:
CARLA version: 
Platform/OS: Ubuntu
Problem you have experienced: How to generate dataset labels with the same format as nuscenes
What you expected to happen: The related source code

wjmcidi, 2024-03-28:
> CARLA version: Platform/OS: Ubuntu Problem you have experienced: How to generate dataset labels with the same format as nuscenes What you expected to happen: The related source code

do you find some examples? i also look forward these example

MIKE-GUO233, 2024-04-02:
is this problem has been solved? I want some examples too! thanks.

generalchan825, 2024-04-08:
Hi Bruce, have you come up with any solution @WangzcBruce ?

RobotiX101, 2024-08-02:
still struggle with this, maybe I'll try bench2drive

yyongjae, 2025-01-08:
is there anybody who solve this problem?