carla-simulator/carla #6789: adding new props

lsangreg, 2023-09-20:
Hello everyone. 
I have been trying adding new props in carla using UE4 and https://carla.readthedocs.io/en/0.9.12/tuto_A_add_props/#ingestion-in-a-build-from-source this tutorial.
This is the json: 
```json
{
  "maps": [],
  "props": [
    {
      "name": "Roller",
      "size": "medium",
      "source": "./Props/Prop01/SoilRoller.fbx",
      "tag": "Static"
    }]
}
```
My package is called Roller, when I import it using `make import` I can see it in  UE4 and I have a folder structure like this:
├── Roller
│   ├── Config
│   │   └── Roller.Package.json
│   ├── Maps
│   ├── Static
│   │   ├── Static
│   │   │   ├── Roller
│   │   │   │   └── SoilRoller_SoilRoller.uasset
│   │   │   │   └── Material.uasset

When it comes to export the package as standalone for CARLA, I use command `make package ARGS="--packages=Roller"`  and this is what I obtain, in the tar.gz file:

 ├── Roller_0.9.12-dirty
│   ├── CarlaUE4
│   │   ├── Content
│   │   │   ├── Carla
│   │   │   │   ├── Blueprints
│   │   │   │   ├── Static 
│   │   │   ├── Roller
│   │   │   │   ├── Config
│   ├── Engine

So inside the Roller i have just config, without the Static folder with the uassets and the materials.

I also tried adding these files manually, so I imported them in carla but then I could not find them in the blueprints, as static actor to spawn, which from what I understood,  should be possible.

_Originally posted by @lsangreg in https://github.com/carla-simulator/carla/discussions/6788_

HeyLittleBlack, 2023-09-26:
I had a similar problem, which I finally solved by changing the name of prop in ueeditor. When I import my model into CarlaUe, the name of the model mesh is automatically added after it. I just delete any extra names. For example, my model name is abc and my mesh name is abc, which becomes abc_abc when imported.

lsangreg, 2023-09-27:
@HeyLittleBlack  Thank you very much, that solved the issue!

closed 2023-09-27 by lsangreg