carla-simulator/carla #6795: py--3   Problem:  building PythonAPI

RGB3Q, 2023-09-22:
problem encountered while building carla PythonAPI, anyone has any idea ? 

 1>Checking Build System
  osm2odr.vcxproj -> D:\Carla\Build\osm2odr-visualstudio\src\Release\osm2odr.lib
  Building Custom Rule D:/Carla/Build/om2odr-source/CMakeLists.txt
  1>
  -- Install configuration: "Release"
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "D:\Carla\PythonAPI\carla\dependencies\"
-[BuildPythonAPI]: [Batch params]: --py3
Building Python API for Python 3.
Traceback (most recent call last):
  File "<string>", line 6, in <module>
  File "__main__.py", line 126, in <module>
  File "__main__py__.py", line 60, in <module>
IOError: [Errno 2] No such file or directory: '-3'

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "D:\Carla\PythonAPI\carla\dist"!

xavisolesoft, 2024-04-19:
We removed the usage of `py -3` command on Windows, currently we use always the `python` command.

Was this fixed your issue?

xavisolesoft, 2024-04-26:
I close the issue for inactivity