carla-simulator/carla #6799: RuntimeError: time-out of 5000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000

Youngzipper, 2023-09-27:


Youngzipper, 2023-09-27:
I'm using vscode to remotely connect to a linux server, doing some experiments with controlling carla using python. But encountered an error: RuntimeError: time-out of 5000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000.
My python codes that encounter this error are as follows:
`try:
            self.world = self.client.get_world()
        except RuntimeError:
            # use subprocess to open CARLA
            subprocess.Popen(self.carla_path)
            time.sleep(5)
            # create a CARLA client
            self.client = carla.Client('localhost', 2000)
            self.client.set_timeout(5.0)

            self.world = self.client.load_world("Town04")

        self.spectator = self.world.get_spectator()`

Youngzipper, 2023-10-03:
does anyone encounter the same problem

nusstu0019, 2023-10-06:
yes, I am facing the same problem

Youngzipper, 2023-10-06:
> yes, I am facing the same problem

Have you got a solution? Or can you restart CARLA when crashes? I added the codes "subprocess.Popen(carla_path)" in my Python script to restart CARLA but failed.

Youngzipper, 2023-10-06:
> yes, I am facing the same problem

Oh sorry, more details I will provide. I encountered the crash when my RL codes reached about 35 epoches (40 mins).
Complete error information is :
(base) working_station:~/Carla$ ./CarlaUE4.sh 
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=196744 
Malloc Size=131160 LargeMemoryPoolOffset=327928 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)

nusstu0019, 2023-10-06:
I am sorry, I am searching for solution too.

Youngzipper, 2023-10-06:
> I am sorry, I am searching for solution too.
@
Maybe we can have each other emails and communicate on CARLA later. My email is yzy.atom@gmail.com

demirciogl, 2023-10-09:
carlaue4.exe may already be running in the background. End the carlaue4.exe task in Task Manager (Ctrl+Alt+Del) and try to run the script again.

slDeng1003, 2024-10-16:
@Youngzipper  @LandoFan 
Hi, I encounter the same problem. 
My hardware is Ubuntu 20.4, Anaconda virtual environment for Python 3.7, Carla 0.9.11, with SSH remote connection to the computer.
Here is my solution:
Step1: Move to the root directory of carla:
`cd carla`
Step2: Run carla server before call it in your code:
Importantly: you should explictly assign a port for it (see **command-line options** in the official document (search for "command-line" in the page) https://carla.readthedocs.io/en/latest/start_quickstart/#running-carla
`./CarlaUE4.sh -carla-rpc-port=3000`
Step3: Run your code by:
`cd path_of_your_code_dir`
`python yourcode.py`
Do not close/kill carla server until your code ends.