carla-simulator/carla #6804: Carla start with segmentation fault

AlgorithmVoyager, 2023-09-30:


AlgorithmVoyager, 2023-09-30:
I find when i run the command， the GPU wasn't effect，
Sat Sep 30 10:14:42 2023       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.104.05             Driver Version: 535.104.05   CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce RTX 3060        Off | 00000000:2B:00.0  On |                  N/A |
|  0%   43C    P5              23W / 170W |    159MiB /  8192MiB |     35%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A      2074      G   /usr/lib/xorg/Xorg                          154MiB |
+---------------------------------------------------------------------------------------+

AOOOOOA, 2023-11-29:
> ## CARLA version: 0.9.14
> Platform/OS: Ubuntu22.04
> Problem you have experienced: ./CarlaUE4.sh with black screen
> ## 4.26.2-0+++UE4+Release-4.26 522 0
> Disabling core dumps.
> LowLevelFatalError [File:Unknown] [Line: 1214]
> GameThread timed out waiting for RenderThread after 60.00 secs
> Signal 11 caught.
> Malloc Size=65538 LargeMemoryPoolOffset=65554
> CommonUnixCrashHandler: Signal=11
> Malloc Size=131160 LargeMemoryPoolOffset=196744
> Malloc Size=131160 LargeMemoryPoolOffset=327928
> Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
> Segmentation fault
> Other information (documentation you consulted, workarounds you tried): add -opengl or other config， not works

Have you fix this?

mz7790, 2024-01-27:
CARLA version: 0.9.14
CARLA version: 0.9.13
Platform/OS: Ubuntu20.04

Problem you have experienced: ./CarlaUE4.sh with black screen

4.26.2-0+++UE4+Release-4.26 522 0
LowLevelFatalError [File:Unknown] [Line: 1214] 
GameThread timed out waiting for RenderThread after 60.00 secs
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=196744 
Malloc Size=131160 LargeMemoryPoolOffset=327928 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault

add -opengl or --vulkan ，no works

How to fix this？

AuderySun, 2024-02-16:
@mz7790 
Hi, have you been able to fix this error? I am meeting the same error.

Morizb, 2024-04-07:
Hi, have you been able to fix this error? I am meeting the same error.

> I find when i run the command， the GPU wasn't effect， Sat Sep 30 10:14:42 2023 +---------------------------------------------------------------------------------------+ | NVIDIA-SMI 535.104.05 [...]

mz7790, 2024-04-07:
Hello, I tried to resolve the error you encountered but was unsuccessful. Interestingly, after switching to a different computer, the error disappeared. I hope this information is helpful to you. If you need further assistance or would like to discuss more, please feel free to let me know. Thank you!

| |
梅震
|
|
***@***.***
|
---- Replied Message ----
| From | ***@***.***> |
| Date | 4/7/2024 16:41 |
| To | ***@***.***> |
| Cc | ***@***.***>,
***@***.***> |
| Subject | Re: [carla-simulator/carla] Carla start with segmentation fault (Issue #6804) |

Hi, have you been able to fix this error? I am meeting the same error.

I find when i run the command， the GPU wasn't effect， Sat Sep 30 10:14:42 2023 +---------------------------------------------------------------------------------------+ | NVIDIA-SMI 535.104.05 Driver Version: 535.104.05 CUDA Version: 12.2 | |-----------------------------------------+----------------------+----------------------+ | GPU Name Persistence-M | Bus-Id Disp.A | Volatile Uncorr. ECC | | Fan Temp Perf Pwr:Usage/Cap | Memory-Usage | GPU-Util Compute M. | | | | MIG M. | |=========================================+======================+======================| | 0 NVIDIA GeForce RTX 3060 Off | 00000000:2B:00.0 On | N/A | | 0% 43C P5 23W / 170W | 159MiB / 8192MiB | 35% Default | | | | N/A | +-----------------------------------------+----------------------+----------------------+

+---------------------------------------------------------------------------------------+ | Processes: | | GPU GI CI PID Type Process name GPU Memory | | ID ID Usage | |=======================================================================================| | 0 N/A N/A 2074 G /usr/lib/xorg/Xorg 154MiB | +---------------------------------------------------------------------------------------+

MadhavanTR, 2024-07-26:
It works on native Windows/Ubuntu, but I get this error when I try on Ubuntu 22.04 using WSL2.

Double-checked for driver/cuda-toolkit toolkit issues

Other GUI apps run fine.

Currently, I'm running the server on native Windows, and ros nodes on Ubuntu on WSL.

commenting for updates on how to fix this issue....

YiHuang108, 2024-08-14:
got the same error in CARLA version: 0.9.15