carla-simulator/carla #6805: the problem of installing carla

nusstu0019, 2023-09-30:
Thanks for contributing to CARLA!

If you are reporting an issue, please use the following outline:

CARLA version:
Platform/OS: Ubuntu 20.04
Problem you have experienced: when I install software requirements. "sudo add-apt-repository ppa:ubuntu-toolchain-r/test". The problem happen:“http://apt.llvm.org/focal llvm-toolchain-focal-8 Release” do not have release file.
What you expected to happen: How should I solve the problem.
[image]

chungcode0218, 2024-01-24:
hello, I have a same problem, did you fixed it ?

nusstu0019, 2024-01-24:
I am sorry for forgetting how to solve it. It seems to be that I find the relavant file and download it by hand.

chungcode0218, 2024-01-24:
OK~thanks!