carla-simulator/carla #6812: Run leaderboard_evaluator with human agent, but shows incorrect window

Imperialpiggy, 2023-10-08:
I am a beginner in Carla and I am trying to run leaderboard_evaluator.py with options 
args_list = ['--routes', 'D:/workspace/Carla/leaderboard-master/data/routes_devtest.xml', '--route-id', '2', '--scenarios', 'D:/workspace/Carla/leaderboard-master/data/all_towns_traffic_scenarios_public.json', '--agent', 'D:/workspace/Carla/leaderboard-master/leaderboard/autoagents/human_agent.py', '--debug', '1']. But the human agent windows cannot show properly:
[image]
Meanwhile, the CarlaUE4 main windows displays normally.
[image]
Carla 0.9.14.
Thanks in advance for your help and attention.

Imperialpiggy, 2023-10-09:
It seems that the problem is caused by the use of a wrong version. I fixed it by downloading the leaderboard-2.0 version instead of leaderboard-master branch.

gezp, 2023-10-09:
it seems a bug for version 0.9.14  in windows, refer to https://github.com/carla-simulator/carla/issues/6085.