carla-simulator/carla #6815: Build Errors When Compiling Python API on Linux

i-kj, 2023-10-09:
Hello,

I am encountering build errors when trying to compile the Python API on Linux(Ubuntu). I followed the instructions on the CARLA documentation, and I have attempted both make PythonAPI and make PythonAPI ARGS="--python-version=3.7", yet I am facing issues with missing files and warnings regarding deprecated builtins.

Here is a snippet of the error message I am getting:

...failed updating 2 targets...
...skipped 15 targets...
...updated 261 targets...
make: *** [Util/BuildTools/Linux.mk:137: setup] Error 1

I am using Python 2.7.18 and the latest version of Carla, and I have ensured that all dependencies are correctly installed. Any assistance in resolving these build issues would be greatly appreciated.

Thank you.
-->

Willis814, 2023-10-10:
I also encountered this problem, but I was using python3.8.10 and all environments were installed. If anyone can help figure this out I would be very grateful.

i-kj, 2023-10-10:
Did you find any solution for it?

Willis814, 2023-10-10:
Yeah，I found some dependencies that were not installed and installed them,then I suceesfully built it. Maybe you could rebuild Carla.

i-kj, 2023-10-10:
Yep. I also installed the dependencies one by one again just to make sure all installs are successful and it worked. 

I found an error while installing 

pip3 install --user -Iv setuptools==47.3.1 

Had to install:

pip install --user testresources

to fix it.

closed 2023-10-10 by i-kj

Willis814, 2023-10-10:
https://carla.readthedocs.io/en/latest/start_quickstart/#install-client-library
Maybe you can follow the official instructions carefully step by step.I've literally had the same problem before, but laterly I rebuilt it and succeed.

i-kj, 2023-10-12:
This is quick start installation. Not the developer build. Did you install the linux build: https://carla.readthedocs.io/en/latest/build_linux/ or https://carla.readthedocs.io/en/latest/start_quickstart/#install-client-library ? 

And did you face any version mistmatch issue while running API?

WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.13 
WARNING: Simulator API version  = 0.9.14-6-g43b5e7064 
Segmentation fault (core dumped)