carla-simulator/carla #6822: No waypoint on the onramp road is fatal error to the vehicle's autopilot!!!

Liuzy0908, 2023-10-11:
# CARLA version: 0.9.13
# Platform/OS: Ubuntu18.04
# Problem you have experienced:

When read a map of an onramp scenario from OpenDrive(.xodr file), the road type at the onramp is carla.LaneType.Onramp, and the road type of the main road is carla.LaneType.Driving.

---
## Fristly:
OpenDrive(.xodr file): the road type at the onramp is carla.LaneType.Onramp
```
<lane id="-2" type="onRamp" level="false">
```
```world = client.generate_opendrive_world(xodr_file.read())
all_spawn_points = world.get_map().get_spawn_points()
for idx, waypoint_location in enumerate(all_default_spawn):
    world.debug.draw_string(waypoint_location.location, str(idx), life_time=100)
```

As you can see, no waypoint is generated on the on-ramp!

---
## Secondly:
Change the road type at the onramp in xodr: carla.LaneType.Onramp -> carla.LaneType.Driving
```
<lane id="-2" type="driving" level="false">
```
```
world = client.generate_opendrive_world(xodr_file.read())
all_spawn_points = world.get_map().get_spawn_points()
for idx, waypoint_location in enumerate(all_default_spawn):
    world.debug.draw_string(waypoint_location.location, str(idx), life_time=100)
```
As you can see, waypoints is generated on the onramp road!

---
# What you expected to happen:

Why does carla only support generating waypoints on roads of carla.LaneType.Driving type?

No waypoint on the onramp road is fatal error to the vehicle's autopilot!!!

Please tell me how to generate waypoints on the onramp road at carla.LaneType.Onramp type. Thanks a lot.

Your sincerely!

Kait0, 2025-08-21:
Also, seeing the issue that the OpenDrive tags onRamp and offramp are not usable in CARLA (and not labelled in the maps), as well as some others like Exit and Entry.
One issue that I see is the default map.get_waypoint(lane_type=carla.LaneType.Driving) defaults to carla.LaneType.Driving.
This is used in various places throughout the code, such as the A* planner.
Using the correct offramp and on ramp labels would lead to broken graphs there.

map.get_waypoint( does allow for multiple types, so one could replaces all instances with 
`map.get_waypoint( lane_type=(carla.LaneType.Driving |carla.LaneType.OffRamp |carla.LaneType.OnRamp) ).`

The A* is part of the CARLA distribution, so it would be better to be fixed internally.

Another problem is that [junction.get_waypoints()](https://github.com/carla-simulator/carla/blob/dbdaaf3719415abc2291af06134e6f84a88c4a54/LibCarla/source/carla/client/Junction.h#L34) does not allow for multiple types because it takes an enum directly and does not convert it to int like [map.get_waypoint](https://github.com/carla-simulator/carla/blob/dbdaaf3719415abc2291af06134e6f84a88c4a54/LibCarla/source/carla/road/Map.h#L56), so the workaround does not work here.

Fixing junction.get_waypoints( would be the easier solution and then updating the default of both functions to an or off all drivable types.

Or in general, it would be nice if there were an explicit notion of drivable lane_type.

M-Colley, 2026-10-26:
Draft PR #10021 (`ue58-dev`) and #10005 (`ue4-dev`) clarify the documented driving-lane filter used by these waypoint APIs. These are documentation changes: they do not add OnRamp waypoint generation or fix autopilot support for OnRamp lanes. Both drafts currently have merge conflicts.