carla-simulator/carla #6832: When I add a custom 4 wheels vehicle in carla, the front wheels sink into the ground, but the rear wheels can be detect normally.

eggman-1024, 2023-10-15:
the vehicle`s size is small, the length of the vehicle is about 98 cm, and the width is about46.5cm, the radius of the wheels is 16.5 cm.
please look at the vedio.

Uploading 072d52b2a26ab01a97bfe66b3f349f07.mp4…

i-kj, 2023-10-16:
You can check this tutorial https://www.youtube.com/watch?v=0F3ugwkISGk

In the Class Defaults panel, set Collision Mesh to Wheel_Shape. Omitting this step will cause the vehicle wheels to sink into the ground.

eggman-1024, 2023-10-16:
> You can check this tutorial https://www.youtube.com/watch?v=0F3ugwkISGk
> 
> In the Class Defaults panel, set Collision Mesh to Wheel_Shape. Omitting this step will cause the vehicle wheels to sink into the ground.

I have already set the Collision Mesh to Wheel_Shape, but it only work for the rear wheels

AinaRoca, 2024-04-19:
If the wheels are smaller than a real car, you should create a different Wheel_Shape with a size that fits your car's wheels, and assign it to each of the wheels.
When you put the new Wheel_Shape on the collision mesh of the wheels and compile, be careful with the tire config, it may disappear. If it does, assign again the CommonTireConfig and compile again.

Then you have to make sure that the center of mass is in the correct place. Drag the car blueprint to the scene, and to see it, go to "Show" (in the viewport) -> Advanced -> check "mass properties".
If it's not in the center of the car, you have to move it (it may be the reason why the front wheels go under the road). Select the car on the  world outliner, select the mesh on details, and search for “Center of mass offset”. X and Y should be in 0, then move the Z down to the lowest part of the chassis.
Copy those values and put them on the car blueprint (open blueprint, select mesh, search on details Center of mass Offset and paste the values)
And it should work.

AinaRoca, 2024-04-26:
I consider this issue solved, so I'm going to close it.
If there is any doubt or problem, feel free to open it again.

closed 2024-04-26 by AinaRoca

abizovnuralem, 2024-08-10:
Configure the wheel blueprints.

In your vehicle blueprint folder, open all four of the wheel blueprints.
In the Class Defaults panel, set Collision Mesh to Wheel_Shape. Omitting this step will cause the vehicle wheels to sink into the ground.