carla-simulator/carla #6836: roadrunner plugins installation error, the version on  mathworks website is latest, does anyone have the previous one?

zyvonne, 2023-10-17:


zyvonne, 2023-11-01:
@bobpan94 , I gave up plugin installation and follow the [tuto_M_add_map_source](https://carla.readthedocs.io/en/0.9.14/tuto_M_add_map_source/) and works.
1. put the `.xodr` and `.fbx` in the Import Folder Path and set `.json` file
2. `make import ARGS="--package=Package --no-carla-materials"`
3. success
remember you need to draw closed-loop roads otherwise error when simulation. For me, it happend :)

closed 2023-11-01 by zyvonne

JoFeGoOr, 2023-11-02:
i had the same problem, i tried the "make import" method but there is no new folder in "<carla_root>/Unreal/CarlaUE4/Content", does anyone know where i can find the new map?

this is the output of make import ARGS="--package=Tutorial --no-carla-materials"

C:\carla>make import ARGS="--package=Tutorial --no-carla-materials"
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --package=Tutorial --no-carla-materials
-[Setup]: Asynchronous jobs:  20
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Install directory:  "C:\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "C:\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "C:\carla\Build\" --zlib-install-dir "C:\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "C:\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "C:\carla\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_gtest]: A Google Test installation already exists.
    -[install_gtest]: Delete "C:\carla\Build\gtest-install\" if you want to force a rebuild.
    -[install_gtest]: Exiting...
-[Setup]: Installing "Recast & Detour"...
    -[install_recast]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_recast]: A "Recast & Detour" installation already exists.
    -[install_recast]: Delete "C:\carla\Build\recast-0b13b0-install\" if you want to force a rebuild.
    -[install_recast]: Exiting...
-[Setup]: Installing Boost...
    -[install_boost]: [Batch params]: --build-dir "C:\carla\Build\" --toolset msvc-14.2 --version 1.80.0 -j 20
    -[install_boost]: A Boost installation already exists.
    -[install_boost]: Delete "C:\carla\Build\boost-1.80.0-install\" if you want to force a rebuild.
    -[install_boost]: Exiting...
-[Setup]: Installing Xercesc...
    -[install_xercesc]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_xercesc]: A xerces installation already exists.
    -[install_xercesc]: Delete "C:\carla\Build\xerces-c-3.2.3-install\" if you want to force a rebuild.
    -[install_xercesc]: Exiting...
        1 archivo(s) copiado(s).
-[Setup]: Installing Sqlite3
    -[install_sqlite3]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_sqlite3]: A Sqlite3 installation already exists.
    -[install_sqlite3]: Delete "C:\carla\Build\sqlite3-install" if you want to force a rebuild.
    -[install_sqlite3]: Exiting...
        1 archivo(s) copiado(s).
-[Setup]: Installing PROJ
    -[install_proj]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_proj]: A PROJ installation already exists.
    -[install_proj]: Delete "C:\carla\Build\proj-install" if you want to force a rebuild.
    -[install_proj]: Exiting...
        1 archivo(s) copiado(s).
-[Setup]: Installing Eigen
    -[install_eigen]: [Batch params]: --build-dir "C:\carla\Build\"
0 archivo(s) copiado(s)
0 archivo(s) copiado(s)
        1 archivo(s) copiado(s).
        1 archivo(s) copiado(s).

    -[install_eigen]: Eigen has been successfully installed in "C:\carla\Build\eigen-install"!
    -[install_eigen]: Exiting...
-[Setup]: Creating "CMakeLists.txt.in"...
-[Setup]:
   ###########
   # SUCCESS #
   ###########

   IMPORTANT!

   All the CARLA library dependences should be installed now.
   (You can remove all "*-src" folders in C:\carla\Build\ directory)

   You only need the ASSET PACK with all the meshes and textures.

   This script provides the assets for CARLA 0.9.14:
   You can download the assets from here:

       http://carla-assets.s3.amazonaws.com/20221201_5ec9328.tar.gz

   Unzip it in the "C:\carla\Unreal\CarlaUE4\Content\Carla\" folder.
   If you want another version, search it in C:\carla\Util\ContentVersions.txt.

-[BuildLibCarla]: [Batch params]: --server
-- Build debug:   ON
-- Build release: ON
-- Build test:    ON
-- Configuring done (0.0s)
-- Generating done (0.1s)
-- Build files have been written to: C:/carla/Build/libcarla-visualstudio
Microsoft (R) Build Engine versión 16.11.2+f32259642 para .NET Framework
Copyright (C) Microsoft Corporation. Todos los derechos reservados.

  1>Checking Build System
  carla_server.vcxproj -> C:\carla\Build\libcarla-visualstudio\LibCarla\cmake\server\Release\carla_server.lib
  carla_server_debug.vcxproj -> C:\carla\Build\libcarla-visualstudio\LibCarla\cmake\server\Release\carla_server_debug.lib
  1>
  -- Install configuration: "Release"
-[BuildLibCarla]: LibCarla server has been successfully installed in "C:\carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\"!

[main 2023-11-02T18:09:15.188Z] update#setState idle
[main 2023-11-02T18:09:21.589Z] Extension host with pid 15492 exited with code: 0, signal: unknown.