carla-simulator/carla #6840: Generating buildings and other features present in the osm file to include in xodr file.

Maheshmirje, 2023-10-19:


Maheshmirje, 2023-10-19:
I need to generate buildings and other features present in the osm file to include in the xodr file. There's a github repo that does this: https://github.com/JHMeusener/osm2xodr. This only generates the road networks and produces a mesh for the road. Can we generate the buildings present in the osm file automatically via a python script or do we need to manually build using softwares such as unreal engine?

i am totally new to unreal engine so i dont know how to use it. I thought of using osmnx (github link:https://osmnx.readthedocs.io/en/stable/getting-started.html) to include the buildings and other features present in the osm file but i dont know how to use it too. Please drop your help.

Blyron, 2024-09-04:
We have digital twin tool for this process, you can drop your osm file to the tool to generate buildings and roads

closed 2024-09-04 by Blyron