carla-simulator/carla #6847: [Open3d Semantic Lidar Visualization] index out of bounds

yyxssm, 2023-10-22:
When running open3d visualization example code in Carla 0.9.14 version, **IndexError** is occurred. The error happens in  [https://github.com/carla-simulator/carla/blob/0.9.14/PythonAPI/examples/open3d_lidar.py#L108](url)

The error is:
```
Traceback (most recent call last):
  File "PythonAPI/examples/open3d_lidar.py", line 198, in <lambda>
    lidar.listen(lambda data: semantic_lidar_callback(data, point_list))
  File "PythonAPI/examples/open3d_lidar.py", line 113, in semantic_lidar_callback
    int_color = LABEL_COLORS[labels]
IndexError: index 25 is out of bounds for axis 0 with size 23
```

I think it is because the color mapping in [https://github.com/carla-simulator/carla/blob/0.9.14/PythonAPI/examples/open3d_lidar.py#L34](url) is not updated. Semantic classes outnumbering visualization colors causes the error.

I do not know the specific **mapping of ObjTag and semantic label**, could anyone tell my how to find the labl mapping? Thanks a lot!

DeepDuke, 2024-01-21:
same problem. What is the total semantic label?

DeepDuke, 2024-01-21:
The `open3d_lidar.py` should work for `version <= 0.9.13` , where the semantic label is defined as:
https://github.com/carla-simulator/carla/blob/0.9.13/LibCarla/source/carla/rpc/ObjectLabel.h#L16-L41

But for later version like `0.9.14` and `0.9.15`, it changed as:
https://sourcegraph.com/github.com/carla-simulator/carla@d7b45c1e159e6d13296f7a3a4e8b13e6c2d62c18/-/blob/LibCarla/source/carla/rpc/ObjectLabel.h#L16-L50

yyxssm, 2024-01-21:
> The `open3d_lidar.py` should work for `version <= 0.9.13` , where the semantic label is defined as 23 labels:
> 
> https://github.com/carla-simulator/carla/blob/a1b37f7f1cf34b0f6f77973c469926ea368d1507/LibCarla/source/carla/rpc/ObjectLabel.h#L16-L41
> 
> 
> But for later version like `0.9.14` and `0.9.15`, it changed as 29 labels:
> https://github.com/carla-simulator/carla/blob/d7b45c1e159e6d13296f7a3a4e8b13e6c2d62c18/LibCarla/source/carla/rpc/ObjectLabel.h#L16-L50
> 
> I guess this is the difference. But the document and examples code haven't been updated yet...

Thank you very much!

PR #8926 (2025-07-02): The open3d_lidar.py LABEL_COLORS have been updated.

#### Description

The LABEL_COLORS have been updated. The `open3d_lidar.py` semantic lidar
used to return IndexError out of bound, since more semantic labels have
been added to Carla. However, the example script was not updated. 

Fixes #6847 

#### Where has this been tested?

  * **Platform(s):** Ubuntu 24.04
  * **Python version(s):** 3.7
  * **Unreal Engine version(s):** (only python script was updated).

- - -
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/8926)