carla-simulator/carla #6848: carla.WalkerBoneControl() -- AttributeError: module 'carla' has no attribute 'WalkerBoneControl'

smogrewvic, 2023-10-23:
Following the PythonAPI documentation on controlling walker bones found here:

https://carla.readthedocs.io/en/latest/tuto_G_control_walker_skeletons/

I run into the following issue when copying the example code listed above:
control = carla.WalkerBoneControl()

 -- AttributeError: module 'carla' has no attribute 'WalkerBoneControl'

Has anyone run into this error or can provide a small code snippet on how to implement bone movement in walkers?

 I'm trying to make a walker turn their head to look at a target position, where the head angle is obtained by calculating the relative angle between the walker and the target, then applying the appropriate yaw angle in  first_tuple = ('crl_Head__C', carla.Transform(rotation=carla.Rotation(yaw = angle_to_target))) 

Thanks for any suggestions or solutions

Ezharjan, 2024-01-29:
Encountering the same problem.

jorge2najo, 2024-08-20:
Same problem.

M-Colley, 2026-10-26:
Draft PR #10019 corrects the obsolete `ue4-dev` walker-bone documentation to use `WalkerBoneControlIn` and `set_bones`, with an executable documentation-example regression. It currently has merge conflicts. Live walker animation validation remains unrun.