carla-simulator/carla #6853: Recast&Detour commit hash error - Could not parse object

Dragonking3x, 2023-10-25:
CARLA version: 0.9.14 (git close version)
Platform/OS: ubuntu 20.04

[image]

you can test it with 
```
git clone https://github.com/carla-simulator/recastnavigation.git recast-0b13b0-10-source
cd recast-0b13b0-10-source
git reset --hard 0b13b0d288ac96fdc5347ee38299511c6e9400db
```

using another commit hash works
```
git reset --hard af6d4758ffae1d192de50b9c6217ee58211baa6a
```

karenjennyli, 2023-10-25:
I got this error as well and made these changes:
Change `Util/BuildTools/Setup.sh:334` to `RECAST_HASH=af6d47`
Change `Util/BuildTools/Setup.sh:335` to `RECAST_COMMIT=af6d4758ffae1d192de50b9c6217ee58211baa6a`

I'm now getting this error that some file doesn't exist.
```
-- Build files have been written to: /home/karenli/carla/Build/recast-af6d47-c12-build
[39/39] Linking CXX executable Tests/Tests
[0/1] Install the project...
-- Install configuration: ""
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/include/recastnavigation/version.h
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/lib/pkgconfig/recastnavigation.pc
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/lib/cmake/recastnavigation/recastnavigation-targets.cmake
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/lib/cmake/recastnavigation/recastnavigation-targets-noconfig.cmake
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/lib/cmake/recastnavigation/recastnavigation-config.cmake
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/lib/cmake/recastnavigation/recastnavigation-config-version.cmake
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/lib/libDebugUtils.a
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/include/recastnavigation/DebugDraw.h
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/include/recastnavigation/DetourDebugDraw.h
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/include/recastnavigation/RecastDebugDraw.h
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/include/recastnavigation/RecastDump.h
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/lib/libDetour.a
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/include/recastnavigation/DetourAlloc.h
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/include/recastnavigation/DetourAssert.h
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/include/recastnavigation/DetourCommon.h
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/include/recastnavigation/DetourMath.h
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/include/recastnavigation/DetourNavMesh.h
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/include/recastnavigation/DetourNavMeshBuilder.h
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/include/recastnavigation/DetourNavMeshQuery.h
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/include/recastnavigation/DetourNode.h
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/include/recastnavigation/DetourStatus.h
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/lib/libDetourCrowd.a
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/include/recastnavigation/DetourCrowd.h
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/include/recastnavigation/DetourLocalBoundary.h
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/include/recastnavigation/DetourObstacleAvoidance.h
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/include/recastnavigation/DetourPathCorridor.h
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/include/recastnavigation/DetourPathQueue.h
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/include/recastnavigation/DetourProximityGrid.h
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/lib/libDetourTileCache.a
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/include/recastnavigation/DetourTileCache.h
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/include/recastnavigation/DetourTileCacheBuilder.h
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/lib/libRecast.a
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/include/recastnavigation/Recast.h
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/include/recastnavigation/RecastAlloc.h
-- Installing: /home/karenli/carla/Build/recast-af6d47-c12-install/include/recastnavigation/RecastAssert.h
mv: cannot stat '/home/karenli/carla/Build/recast-af6d47-c12-install/include/*h': No such file or directory
make: *** [Util/BuildTools/Linux.mk:137: setup] Error 1
```

dreamfly-fly, 2023-10-26:
注释脚本“carla/Util/BuildTools/Setup.sh”中“git reset --hard ${RECAST_COMMIT}”，大概在356行左右，and try again。

happyman000009, 2023-10-26:
> 注释脚本“carla/Util/BuildTools/Setup.sh”中“git reset --hard ${RECAST_COMMIT}”，大概在356行左右，and try again。

感谢您的建议，但是还会提示：无法获取'recast-0b13b0-c10-install/bin/RecastBuilder' 的文件状态(stat): 没有那个文件或目录

spyeic, 2023-10-26:
> > 注释脚本“carla/Util/BuildTools/Setup.sh”中“git reset --hard ${RECAST_COMMIT}”，大概在356行左右，and try again。
> 
> 感谢您的建议，但是还会提示：无法获取'recast-0b13b0-c10-install/bin/RecastBuilder' 的文件状态(stat): 没有那个文件或目录

我也有一样的错误, 但是第二次运行的时候就没有这个错误了

dreamfly-fly, 2023-10-26:
> > 注释脚本“carla/Util/BuildTools/Setup.sh”中“git reset --hard ${RECAST_COMMIT}”，大概在356行左右，and try again。
> 
> 感谢您的建议，但是还会提示：无法获取'recast-0b13b0-c10-install/bin/RecastBuilder' 的文件状态(stat): 没有那个文件或目录

1.注释“git reset --hard ${RECAST_COMMIT}”
此时报错：'recast-0b13b0-c10-install/bin/RecastBuilder'不存在
2.在上一步注释代码的位置新增“git checkout 0ef0433b0194dbeadd84632f92ae9eb5a11d58f8”
测试报错‘recast-0b13b0-c10-install/bin/*h’不存在
3.将步骤二'recast-0b13b0-c10-install/bin/RecastBuilder'注释后，重新运行 make PythonAPI 
4.中可能还有和git有关的报错，不过这种报错一般会提示一个命令给你解决，执行后重新make PythonAPI 即可

原因：carla分支最近提交的代码可能存在问题，并且删除了之前0b13b0的提交记录

anrp-tri, 2023-10-26:
I happened to have a copy of the git repo locally so I pushed it up to https://github.com/anrp-tri/recastnavigation/commit/0b13b0d288ac96fdc5347ee38299511c6e9400db - looks like it matches 0ef0433b0194dbeadd84632f92ae9eb5a11d58f8 but with the underlying library brought up to date

Jeff-WP, 2023-10-30:
My team ran into this issue as well and lost some time to debugging it last week.

However, it seems that someone restored the referenced commit and tagged it in the recastnavigation repository at [hash_0b13b0](https://github.com/carla-simulator/recastnavigation/tree/hash_0b13b0). Our build is now working again thanks to the tag. I don't know who pushed it, and I've yet to see any communication about what happened, but this issue may now be resolved.