carla-simulator/carla #6854: Could NOT find SDL2

Willis814, 2023-10-27:


Canio95, 2023-10-31:
I encountered the same problem following the Build Carla on Windows guide. I fixed this by cloning Carla 0.9.14 with the following command instead of the one suggested:

git clone https://github.com/carla-simulator/carla -b 0.9.13 --depth 1

Hope it helps

Willis814, 2023-11-01:
Thank you for your help, but it seems that cmake still cannot find sdl2, even though I have installed sdl2.🥺

Willis814, 2024-01-28:
> Have you already sovled it? I met this problem,too.

I have solved it. Maybe you should download the source file of SDL2 and compile it.

ppfei666, 2024-01-28:
> 我已经解决了。也许您应该下载 SDL2 的源文件并对其进行编译。

How to compile it？

XiaoMing2336, 2024-02-01:
I have solved this question,go to this path and read "carla\Build\recast-src\RecastDemo\Contrib\readme-sdl.txt",you will know what to do, you need unzip your downloaded SDL in contrib as the .txt file says.

XiaoMing2336, 2024-02-03:
> > I have solved this question,go to this path and read "carla\Build\recast-src\RecastDemo\Contrib\readme-sdl.txt",you will know what to do, you need unzip your downloaded SDL in contrib as the .txt file says.
> 
> i'm sorry, which file in SDL should be download and where should it be placed 🥺

 Firstly I download and unzip the [SDL2-devel-2.30.0-VC.zip](https://github.com/libsdl-org/SDL/releases/download/release-2.30.0/SDL2-devel-2.30.0-VC.zip) in "D:\carla\Build\recast-src\RecastDemo\Contrib",then I rename the unziped folder as "SDL",It really works for me.