carla-simulator/carla #6857: How to create a distribution package for a car?

burnout2k, 2023-10-27:
Hello ppl!

I have been designing different maps for Carla for a while now and everything works great when it comes to using
make package ARGS="--packages=map_name" and exporting my map for prebuilt Carla clients.

Now I have created a vehicle and it also works great in Unreal Editor when I run Carla, but...
I have no clue how to use "make package" for this car.
Would like to add it to the collection of cars available when running the prebuilt version of Carla! :)

Any clues?
Cheers

closed 2023-10-29 by burnout2k

atakandag, 2024-02-28:
I am dealing with the issue. Any solution on this would help me a lot! :)

burnout2k, 2024-02-28:
I did not find a solution, just a workaround. 
Add the vehicle blueprint to the map and then use ARGS="--packages=map_name" 
All the vehicle parts will be built as well. You just have to sort them out yourself.