carla-simulator/carla #6860: Destroying collision sensor in asynchrounous mode is very likely to cause server crashing.

Morphlng, 2023-10-30:
# Problem Description
- CARLA version: 0.9.13
- Platform/OS: Windows 11 / Ubuntu 22.04
- Problem you have experienced: When you stop and destroy a collision sensor, it is very likely to crash. The console traceback is related to multi-threading issue, i.e. write to an address that is already freed.
- What you expected to happen: The sensor stop listening and gracefully get destroyed.
- Steps to reproduce: This can be very easy to reproduce in a RL environment. When you reset the episode by destroying the sensor, actors and respawn them, there is a high chance that the server will crash (Note: asynchrounous  mode)
- Other information (documentation you consulted, workarounds you tried): I've upload the crashing log as an attachment
- [CarlaUE4.log](https://github.com/carla-simulator/carla/files/13202731/CarlaUE4.log)

# Solution
I've found that if using synchrounous mode, the problem hardly appears. Currently I will switch to sync mode before destroying the collision sensor, and then switch back to whatever mode the RL environment specified.

Morphlng, 2023-11-06:
- I've tried to teleport things away, doesn't seems to help.
- Toggle to sync mode then back to async mode significantly reduce the rate of crashing. However, in our specific usage, we must stay in async mode during cleanup.

AOOOOOA, 2024-06-07:
> * I've tried to teleport things away, doesn't seems to help.
> * Toggle to sync mode then back to async mode significantly reduce the rate of crashing. However, in our specific usage, we must stay in async mode during cleanup.

Hi, I'm curious about this issue. Could you please provide the code so I can reproduce it? Thanks!