carla-simulator/carla #6863: Custom Vehicle not detected by traffic

Canio95, 2023-10-30:
CARLA version: 0.9.14
Platform/OS: Windows 10

Hi everyone, I want to modify the UE4 project so that I can drive my own vehicle (created with UE default wheeledVehiclePawn and not the Carla Pawn) as player 0 like you would in a normal UE project. But I'd also like to have the traffic piloted by Carla algorythm through the Generate_Traffic.py command. My issue is that the AI doesn't detect my vehicle and crashes into it. I already tried creating the vehicle using the Carla Pawn BP class with its inherited VehicleBound Box etc and adding it to the vehicle factory but the problem persists. My goal is to drive my vehicle from the editor, not from manualcontrol.py . What should I do?

https://github.com/carla-simulator/carla/assets/148270000/818f9d29-61cc-413f-aba7-ff2f54a4456f

gsmario, 2024-05-17:
Controlling your vehicle via editor is not suported for now.
You should use manualcontrol.py to drive your own cars, that way generatetraffic will recognice your vehicle.

Blyron, 2024-05-31:
We currently are not gonna support this. CARLA goal is to help people to collect data using agents using interfaces like ROS2 or Python client. So this is not planned in short term. 

Traffic takes agents in the traffic manager, you could look there in order to your car is taken into account

closed 2024-05-31 by Blyron