carla-simulator/carla #6868: How to use different GPU cards to render to different image sensors? or how to assign a sensor to a certain gpu.

TurtleZhong, 2023-11-02:
Thanks for contributing to CARLA!

If you are reporting an issue, please use the following outline:

CARLA version:
Platform/OS:
Problem you have experienced:

I have 4 GPU, and I wanna render 4 RGB sensors, But I do not know how to assign each GPU to Render 1 RGB image. Is there a python example for a multi gpu setup? I found the new feature in the [release note](https://carla.org/2022/12/23/release-0.9.14/) and here is an picture [image] We are particularly interested in how to assign a sensor to a certain gpu and we couldnt find any information on this. I could run multi-gpu feature follow [this link](https://carla.readthedocs.io/en/latest/adv_multigpu/), but I coult not find some posts about use different GPU cards to render to different image sensors

UPTOLIMIT, 2024-04-11:
It is written [here](https://carla.readthedocs.io/en/latest/adv_multigpu/)
# All the system is transparent to the user, who don't need to know from which server comes the sensor data, he just receives the data directly from a secondary server.
Does it mean that users cannot specify a GPU to render a specific sensor?

TurtleZhong, 2024-04-11:
> It is written [here](https://carla.readthedocs.io/en/latest/adv_multigpu/)
>
> # All the system is transparent to the user, who don't need to know from which server comes the sensor data, he just [...]

I think users can spercify a GPU to render a specific sensor since I check the carla 0.9.14 source codes, but I do not try it.