carla-simulator/carla #6871: Importing maps from roadrunner error. plugin installation faild and make import method does not have visible effects

JoFeGoOr, 2023-11-02:
CARLA version: 0.9.14
Platform/OS: win 11
Problem you have experienced: i cant import a map to carla sim from roadrunner

i followed the ["add a new map"](https://carla.readthedocs.io/en/0.9.14/tuto_M_custom_map_overview/l) guide, in there are 3 methods on how to import a map, i try the plugin method and the make import method.

in the plugin method i get the following output:

```
C:\carla>make launch
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --all
-[Setup]: Asynchronous jobs:  20
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Install directory:  "C:\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "C:\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "C:\carla\Build\" --zlib-install-dir "C:\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "C:\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "C:\carla\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_gtest]: A Google Test installation already exists.
    -[install_gtest]: Delete "C:\carla\Build\gtest-install\" if you want to force a rebuild.
    -[install_gtest]: Exiting...
-[Setup]: Installing "Recast & Detour"...
    -[install_recast]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_recast]: A "Recast & Detour" installation already exists.
    -[install_recast]: Delete "C:\carla\Build\recast-0b13b0-install\" if you want to force a rebuild.
    -[install_recast]: Exiting...
-[Setup]: Installing Boost...
    -[install_boost]: [Batch params]: --build-dir "C:\carla\Build\" --toolset msvc-14.2 --version 1.80.0 -j 20
    -[install_boost]: A Boost installation already exists.
    -[install_boost]: Delete "C:\carla\Build\boost-1.80.0-install\" if you want to force a rebuild.
    -[install_boost]: Exiting...
-[Setup]: Installing Xercesc...
    -[install_xercesc]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_xercesc]: A xerces installation already exists.
    -[install_xercesc]: Delete "C:\carla\Build\xerces-c-3.2.3-install\" if you want to force a rebuild.
    -[install_xercesc]: Exiting...
        1 archivo(s) copiado(s).
-[Setup]: Installing Sqlite3
    -[install_sqlite3]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_sqlite3]: A Sqlite3 installation already exists.
    -[install_sqlite3]: Delete "C:\carla\Build\sqlite3-install" if you want to force a rebuild.
    -[install_sqlite3]: Exiting...
        1 archivo(s) copiado(s).
-[Setup]: Installing PROJ
    -[install_proj]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_proj]: A PROJ installation already exists.
    -[install_proj]: Delete "C:\carla\Build\proj-install" if you want to force a rebuild.
    -[install_proj]: Exiting...
        1 archivo(s) copiado(s).
-[Setup]: Installing Eigen
    -[install_eigen]: [Batch params]: --build-dir "C:\carla\Build\"
0 archivo(s) copiado(s)
0 archivo(s) copiado(s)
        1 archivo(s) copiado(s).
        1 archivo(s) copiado(s).

    -[install_eigen]: Eigen has been successfully installed in "C:\carla\Build\eigen-install"!
[... 217 lines cut ...]
-- Build debug:   ON
-- Build release: ON
-- Build test:    ON
-- Configuring done (0.0s)
-- Generating done (0.1s)
-- Build files have been written to: C:/carla/Build/libcarla-visualstudio
Microsoft (R) Build Engine versión 16.11.2+f32259642 para .NET Framework
Copyright (C) Microsoft Corporation. Todos los derechos reservados.

  1>Checking Build System
  carla_server.vcxproj -> C:\carla\Build\libcarla-visualstudio\LibCarla\cmake\server\Release\carla_server.lib
  carla_server_debug.vcxproj -> C:\carla\Build\libcarla-visualstudio\LibCarla\cmake\server\Release\carla_server_debug.lib
  1>
  -- Install configuration: "Release"
-[BuildLibCarla]: LibCarla server has been successfully installed in "C:\carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\"!

[main 2023-11-02T18:09:15.188Z] update#setState idle
[main 2023-11-02T18:09:21.589Z] Extension host with pid 15492 exited with code: 0, signal: unknown.
```

as you can see, there is no visible error, but there is no new map and no new folder in the "<carla_root>/Unreal/CarlaUE4/Content"

i really have no idea of what can be the problem, i am going to try the 3° method but i really would like to fix any of this methods

thanks for your time

ronyshaji, 2023-11-07:
Hey @JoFeGoOr 
Better go with the manual import method.
It worked for me without any issues.

LBroKerR, 2023-11-07:
@ronyshaji 
Could you tell more about manual import, pls

daohu527, 2024-11-06:
Here [importing-your-road-network-into-carla](https://carla.readthedocs.io/en/latest/tuto_content_authoring_maps/#importing-your-road-network-into-carla)