carla-simulator/carla #6874: deactivate collision between vehicle actors

AbdulkadirAktuerk, 2023-11-05:
**CARLA version:** 0.19.4
**Platform/OS**: Windows

**Problem you have experienced:** 
Currently, in the CARLA simulator, there is no feature that allows for the deactivation of collision between actors, such as vehicles. For my use case, I need to be able to make vehicles "ghost-like" so that they can pass through other vehicles without collision. This feature is crucial for training with a genetic algorithm in a special scenario within the CARLA world, particularly because I aim to run simulations in parallel to increase the speed and efficiency of the training process.

**What you expected to happen:** 
I expected to have an option or a setting within the CARLA simulator that would allow me to toggle the collision feature on and off for specific actors or globally, so that vehicles could ignore collisions with other vehicles when required.

Thanks

lucedes27, 2024-06-05:
I'm also interested in doing this. Has there been any progress?

LyreZ61, 2024-06-05:
Hey @lucedes27,

I'm a friend of Abdulkadir. After some research, we found out that you can create a vehicle in Unreal Engine that ignores other vehicles by deactivating certain attributes. However, this process is very time-consuming. You need to download and install the CARLA source code in Unreal Engine, which alone took us nearly two hours despite having very good hardware.

Our project also required documentation, which was originally written in German. I have translated the steps for you:

------------------------------------------------------------------------------------------------------------------------

### **3.3 Install Unreal Engine and Start Carla Source Code**

**_Note: The installation is very time-consuming and can take between 1-2 hours depending on your hardware._**

Follow this tutorial on the Carla documentation website:

- Follow the instructions on the Carla documentation page:
https://carla.readthedocs.io/en/0.9.14/build_windows/
- Alternatively, you can follow this video tutorial:
https://youtu.be/lLkFA0fPrgs?si=XdNJYJg9rOswuxdr

**Note:**
When you reach the step "Set Unreal Engine environment variable," add the UnrealEngine path for UE4_ROOT and Path.

[image]
[image]

**Note:** To run Carla smoothly in the background, go to UE4's Edit -> Editor Preferences -> Performance and uncheck "Use less CPU when in background."

**Start Carla:** Use **_make launch_** in "x64 Native Tools Command Prompt for VS 2019" to start UE4.

### 3.4 Deactivate Vehicle Collisions
**_3.4.1 Procedure in Unreal Engine:_**

1. Open "x64 Native Tools Command Prompt for VS 2019" and navigate to the Carla SourceCode path.
2. Type **_make launch_** to start UE4 Carla.

[image]
This can take 4-5 minutes. Sometimes you are prompted to open a file; simply close the window and continue waiting.

[image]

3. In UE4, go to the Content Browser path "Content/Carla/Blueprints/Vehicles" and open a vehicle model.

4. Right-click on the "BP_Mercedes" model and duplicate the model, then rename the duplicated model to "BP_Mercedes_NoVehicleCollision."

5. Open the duplicated model and select the Mesh in the Component Menu (top left). Search for Collision under Details and change the Collision Presets from "Vehicle" to "Custom" and set "Vehicle" to "Overlap."

[image]
Click "Compile" at the top and then "Save."

6. Go back to the "Content/Carla/Blueprints/Vehicles" path and open the "VehicleFactory" file.

[image]

8. Click on Vehicles to open the Details panel. (Click on Vehicles on the far left)

[image]

10. Add a new element and label it according to the car as follows:

[image]

Settings vary depending on the vehicle type. It is best to search for your copied vehicle and adopt all the settings. Then change only the model name and class. In my case, the vehicle is named Coupe_NoVehicleCollision and the previously created class is BP_Mercedes_NoVehicleCollision. Then press 'Compile' and 'Save' in the menu.

**_3.4.2 Verification and Packaging:_**

1. Close UE4 and save all changes.
2. Open "x64 Native Tools Command Prompt" again and follow step 2 from 3.4.1.
3. Type **_make package_** to create the package. This can take up to 1 hour. The path to the created build will be displayed.
In my case, it is the path:
C:\Users\Muco\Documents\Carla\CarlaSource\carla\Build\UE4Carla\0.9.15-7-g92111a2cc
4. Start Carla and test if the vehicles have no collisions. Select the appropriate blueprint (e.g., vehicle.mercedes.Coupe_NoVehicleCollision).
[image]
In my case, the blueprint is generated with "vehicle.mercedes.Coupe_NoVehicleCollision."

**_3.4.3 Sharing Blueprints:_**

1. To share blueprints with others, open the self-created Carla build folder (e.g., ...\Carla\CarlaSource\carla\Build\UE4Carla\0.9.15-7-g92111a2cc) and the default Carla folder.
2. Copy the edited files VehicleFactory.uasset and VehicleFactory.uexp as well as the blueprints (e.g., from the Mercedes folder) BP_Mercedes_NoVehicleCollision.uasset and BP_Mercedes_NoVehicleCollision.uexp.
[image]
3. Then replace these files in the default folder to apply the changes and start Carla. (WindowsNoEditor\CarlaUE4\Content\Carla\Blueprints\Vehicles)

[image]

**Important:**
This only works if you copy all changes correctly. For example, if you have modified two vehicles, you must apply all changes to the corresponding files in the standard Carla path.

------------------------------------------------------------------------------------------------------------------------

closed 2024-06-05 by AbdulkadirAktuerk