carla-simulator/carla #6877: make launch frozen while buiding carla from the source code

ElizabethSF, 2023-11-06:
Hello!
I am using Ubuntu 20.04 to build carla0.9.14.

I have successfully installed UE4 and pass the "make PythonAPI"
However, when I am running "make launch", the UE sever successfully starts, but the whole process seem to be struck(frozen). There is no error printed, but the process just refuses to move along.

It freezes with these lines in the terminal:

[2023.11.06-03.42.09:293][  3]LogCore: Done sending crash report for ensure().
[2023.11.06-03.42.09:428][  4]LogAssetRegistry: Asset discovery search completed in 38.4286 seconds
[2023.11.06-03.42.09:431][  4]LogCollectionManager: Fixed up redirectors for 0 collections in 0.000001 seconds (updated 0 objects)
 
Thanks for your attention, I am eager to find solutions...

ronyshaji, 2023-11-07:
@ElizabethSF Hey
Did you already downloaded the Carla assets ?  
if not, please download it and try running make launch.  (run Update.bat at the carla root folder)

ElizabethSF, 2023-11-07:
@ronyshaji Thank you for your response. Yes, I have run ./Update.sh to unpdate carla assets. After my further research, currently I think this is a problem brought by GPU not being found by carla, even though my gpu & nvidia-driver is one of the lateset....

ronyshaji, 2023-11-07:
@ElizabethSF Did you try to open the Unreal Engine directly using unreal>Engine>Binaries>Win64>UE4Editor.exe ?

You can try to open it to check whether the unreal works atleast. But if you do this then you have to build again to use the CARLA.

ElizabethSF, 2023-11-09:
@ronyshaji Thank you again, but this is not my problem. I can successfully open UE on its own. Plus, I am running it on Linux...

ElizabethSF, 2023-11-29:
I found the answer, it turns out that this is not really a problem. What we need is to launch the UE, if it is successfully opened, let it be, whatever warnings or errors are printed.

closed 2023-11-29 by ElizabethSF