carla-simulator/carla #6886: Can not launch carla 0.9.13 on server

AOOOOOA, 2023-11-09:
I tried run Carla 0.9.13 on ubuntu 18.04 server with command  ./CarlaUE4.sh -RenderOffScreen. 

After showing 
" 4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps." 
the program just automatically quit. 

I do not know why since it can be successfully run on other machines with similar computer configurations.

TurtleZhong, 2023-11-15:
check the cuda driver? or try ./CarlaUE4.sh -prefernvidia -RenderOffScreen

AOOOOOA, 2023-11-15:
> check the cuda driver? or try ./CarlaUE4.sh -prefernvidia -RenderOffScreen

Hi thanks for your reply! I checked the CUDA. 

"nvcc -V " returned the version as cuda_11.0
"nvidia-smi" returned the nvidia driver version as 460.32 and the CUDA version as 11.2.

Also I tried the command " ./CarlaUE4.sh -prefernvidia -RenderOffScreen" but nothing changed. 
Btw the system is ubuntu 20.04, I made a typo in the problem description.

Andrea010725, 2024-11-27:
Have you solved this problem? I'm having the same problem.

Andrea010725, 2024-11-27:
I try to run in opengl mode. it finally works !!!
`sudo apt-get install libgl1-mesa-glx libgl1-mesa-dev`
and than run  --no-rendering