carla-simulator/carla #6898: Windows 10: make PythonAPI stops at BuildOSM2ODR, CMAKE build error

HariKrishnaPadala, 2023-11-12:
Hi, 

I am trying to build the Carla 0.9.15 latest source code on Windows 10. 

When I do make PythonAPI in Visual studio 19 x64 command prompt, the process stops 
at BuildOSM2ODR with CMAKE error. 

I think CMAKE cannot generate the visual studio project folder for osm2odr may be because of 
CMAKE configuration issues in Util\BuildTools\BuildOSM2ODR.bat 
There are also no issues with the installation of xerces-c_3.lib while doing make PythonAPI.

Can anyone suggest any solution to the problem?

Thank you.

Lyanyv, 2023-11-13:
same issue for me.
I changed `.\Util\BuildTools\BuildOSM2ODR.bat` line 112 from
``` ps1
    cmake -G %GENERATOR% %PLATFORM%^
```
to
``` ps1
    cmake -G %GENERATOR% -A x64^
```
then the error: `CMake Error: The source directory "C:/build-carla/carla-0.9.15/Build/osm2odr-visualstudio/x64" does not exist.` disappeared, and `make PythonAPI` successed.

hope this helps

sjtuyolee, 2023-11-14:
Great！！！

sjtuyolee, 2023-11-14:
Great!!!
Thanks for your answer @Lyanyv

saitejeswar1, 2023-11-18:
> same issue for me. I changed `.\Util\BuildTools\BuildOSM2ODR.bat` line 112 from
> 
> ```powershell
>     cmake -G %GENERATOR% %PLATFORM%^
> ```
> 
> to
> 
> ```powershell
>     cmake -G %GENERATOR% -A x64^
> ```
> 
> then the error: `CMake Error: The source directory "C:/build-carla/carla-0.9.15/Build/osm2odr-visualstudio/x64" does not exist.` disappeared, and `make PythonAPI` successed.
> 
> hope this helps
Thank you so much @Lyanyv 
I have been struggling to solve this for the past 2 days

vicmen14, 2023-11-27:
**BuildOSM2ODR.bat** script has:

`setlocal enabledelayedexpansion`

so it should use **!variable!** to get access to the variable values. For example, these commands with **setlocal enabledelayedexpansion** at the beggining:

```
set variable=-A x64
echo %variable%
echo !variable!
```

return

```
x64
-A x64
```

so the space in the set operation **"-A x64"** is causing the problems with **setlocal enabledelayedexpansion**. To fix the issue, change this line:

`cmake -G %GENERATOR% %PLATFORM%^`

to 

`cmake -G %GENERATOR% !PLATFORM!^`

vicmen14, 2023-11-27:
jnknkk

vicmen14, 2023-11-27:
**BuildOSM2ODR.bat** has:

`setlocal enabledelayedexpansion`

This means that the variables are not handled as usual. To get access to a variable with **enabledelayedexpansion**, you should use:

`!variable!`

For example, these commands:

```
setlocal enabledelayedexpansion

set variable=-A x64

echo %variable%
echo !variable!
```

return

```
x64
-A x64
```

The space in **"-A x64"** is causing the problem here. To fix this replace line:

cmake -G %GENERATOR% %PLATFORM%^

by 

cmake -G %GENERATOR% !PLATFORM!^

closed 2023-11-28 by HariKrishnaPadala

CMakey, 2023-11-30:
> same issue for me. I changed `.\Util\BuildTools\BuildOSM2ODR.bat` line 112 from
>
> ```powershell
>     cmake -G %GENERATOR% %PLATFORM%^
> ```
>
> to
>
> ```powershell
>     cmake -G %GENERATOR% -A x64^
> ```
>
> then [...]

thanks a lot ! it works!

aminoed, 2024-01-19:
> Does anyone know why the OSM2ODR.bat file on line 112 does not appear for me:
> 
> ```
> cmake -G %GENERATOR% %PLATFORM%^
> ```
> 
> It seems that my OSM2ODR.bat file has been build in a different way than expected... Instead of that, this is my OSM2ODR.bat file:
> 
> #! /bin/bash DOC_STRING="Build OSM2ODR."
> 
> USAGE_STRING=$(cat <<- END Usage: $0 [-h|--help]
> 
> commands
> 
> ```
> [--clean]    Clean intermediate files.
> [--rebuild]  Clean and rebuild both configurations.
> ```
> 
> END )
> 
> REMOVE_INTERMEDIATE=false BUILD_OSM2ODR=false GIT_PULL=true CURRENT_OSM2ODR_COMMIT=1835e1e9538d0778971acc8b19b111834aae7261 OSM2ODR_BRANCH=aaron/defaultsidewalkwidth OSM2ODR_REPO=https://github.com/carla-simulator/sumo.git
> 
> OPTS=`getopt -o h --long help,rebuild,build,clean,carsim,no-pull -n 'parse-options' -- "$@"`
> 
> eval set -- "$OPTS"
> 
> while [[ $# -gt 0 ]]; do case "$1" in --rebuild ) REMOVE_INTERMEDIATE=true; BUILD_OSM2ODR=true; shift ;; --build ) BUILD_OSM2ODR=true; shift ;; --no-pull ) GIT_PULL=false shift ;; --clean ) REMOVE_INTERMEDIATE=true; shift ;; -h | --help ) echo "$DOC_STRING" echo "$USAGE_STRING" exit 1 ;; * ) shift ;; esac done
> 
> source $(dirname "$0")/Environment.sh
> 
> function get_source_code_checksum { local EXCLUDE='_**pycache**_' find "${OSM2ODR_SOURCE_FOLDER}"/* ! -path "${EXCLUDE}" -print0 | sha1sum | awk '{print $1}' }
> 
> if ! { ${REMOVE_INTERMEDIATE} || ${BUILD_OSM2ODR}; }; then fatal_error "Nothing selected to be done." fi
> 
> # ==============================================================================
> # -- Clean intermediate files --------------------------------------------------
> # ==============================================================================
> if ${REMOVE_INTERMEDIATE} ; then
> 
> log "Cleaning intermediate files and folders."
> 
> rm -Rf ${OSM2ODR_BUILD_FOLDER}*
> 
> fi
> 
> # ==============================================================================
> # -- Build library -------------------------------------------------------------
> # ==============================================================================
> if ${BUILD_OSM2ODR} ; then log "Building OSM2ODR."
> 
> # [ ! -d ${OSM2ODR_BUILD_FOLDER} ] && mkdir ${OSM2ODR_BUILD_FOLDER}
> if ${GIT_PULL} ; then if [ ! -d ${OSM2ODR_SOURCE_FOLDER} ] ; then git clone -b ${OSM2ODR_BRANCH} ${OSM2ODR_REPO} ${OSM2ODR_SOURCE_FOLDER} fi cd ${OSM2ODR_SOURCE_FOLDER} git fetch git checkout ${CURRENT_OSM2ODR_COMMIT} fi
> 
> mkdir -p ${OSM2ODR_BUILD_FOLDER} cd ${OSM2ODR_BUILD_FOLDER}
> 
> export CC="$UE4_ROOT/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang" export CXX="$UE4_ROOT/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang++" export PATH="$UE4_ROOT/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin:$PATH"
> 
> cmake ${OSM2ODR_SOURCE_FOLDER} -G "Eclipse CDT4 - Ninja" -DCMAKE_CXX_FLAGS="-stdlib=libstdc++" -DCMAKE_INSTALL_PREFIX=${LIBCARLA_INSTALL_CLIENT_FOLDER} -DPROJ_INCLUDE_DIR=${CARLA_BUILD_FOLDER}/proj-install/include -DPROJ_LIBRARY=${CARLA_BUILD_FOLDER}/proj-install/lib/libproj.a -DXercesC_INCLUDE_DIR=${CARLA_BUILD_FOLDER}/xerces-c-3.2.3-install/include -DXercesC_LIBRARY=${CARLA_BUILD_FOLDER}/xerces-c-3.2.3-install/lib/libxerces-c.a
> 
> ninja osm2odr ninja install
> 
> mkdir -p ${OSM2ODR_SERVER_BUILD_FOLDER} cd ${OSM2ODR_SERVER_BUILD_FOLDER}
> 
> LLVM_BASENAME=llvm-8.0 LLVM_INCLUDE="$UE4_ROOT/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1" LLVM_LIBPATH="$UE4_ROOT/Engine/Source/ThirdParty/Linux/LibCxx/lib/Linux/x86_64-unknown-linux-gnu"
> 
> echo $LLVM_INCLUDE echo $LLVM_LIBPATH
> 
> cmake ${OSM2ODR_SOURCE_FOLDER} -G "Eclipse CDT4 - Ninja" -DCMAKE_CXX_FLAGS="-fPIC -std=c++14 -stdlib=libc++ -I${LLVM_INCLUDE} -L${LLVM_LIBPATH}" -DCMAKE_INSTALL_PREFIX=${LIBCARLA_INSTALL_SERVER_FOLDER} -DPROJ_INCLUDE_DIR=${CARLA_BUILD_FOLDER}/proj-install-server/include -DPROJ_LIBRARY=${CARLA_BUILD_FOLDER}/proj-install-server/lib/libproj.a -DXercesC_INCLUDE_DIR=${CARLA_BUILD_FOLDER}/xerces-c-3.2.3-install-server/include -DXercesC_LIBRARY=${CARLA_BUILD_FOLDER}/xerces-c-3.2.3-install-server/lib/libxerces-c.a
> 
> ninja osm2odr ninja install
> 
> mkdir -p ${OSM2ODR_SERVER_BUILD_FOLDER} cd ${OSM2ODR_SERVER_BUILD_FOLDER}
> 
> LLVM_BASENAME=llvm-8.0 LLVM_INCLUDE="$UE4_ROOT/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1" LLVM_LIBPATH="$UE4_ROOT/Engine/Source/ThirdParty/Linux/LibCxx/lib/Linux/x86_64-unknown-linux-gnu"
> 
> cmake ${OSM2ODR_SOURCE_FOLDER} -G "Eclipse CDT4 - Ninja" -DCMAKE_CXX_FLAGS="-fPIC -std=c++14 -stdlib=libc++ -I${LLVM_INCLUDE} -L${LLVM_LIBPATH}" -DCMAKE_INSTALL_PREFIX=${LIBCARLA_INSTALL_SERVER_FOLDER} -DPROJ_INCLUDE_DIR=${CARLA_BUILD_FOLDER}/proj-install-server/include -DPROJ_LIBRARY=${CARLA_BUILD_FOLDER}/proj-install-server/lib/libproj.a -DXercesC_INCLUDE_DIR=${CARLA_BUILD_FOLDER}/xerces-c-3.2.3-install-server/include -DXercesC_LIBRARY=${CARLA_BUILD_FOLDER}/xerces-c-3.2.3-install-server/lib/libxerces-c.a
> 
> ninja osm2odr ninja install
> 
> fi
> 
> log " OSM2ODR Success!"

That's the "BuildOSM2ODR.sh" File, not the bat file.

juanrodriguezdekra, 2024-01-22:
Yes, you are right. I then proceeded to change the command on line 112 to:
cmake -G %GENERATOR% -A x64^

In the PythonAPI make everything was correct and without incident. 
[image]

But in the make launch I got make: *** [CarlaUE4Editor] Error 6. Apparently without a fatal error, as you can see in the attached image. 
[image]

This is the new point I'm stuck at, in case anyone knows how to get out of here.

juanrodriguezdekra, 2024-01-24:
> Yes, you are right. I then proceeded to change the command on line 112 to: cmake -G %GENERATOR% -A x64^
> 
> In the PythonAPI make everything was correct and without incident. ![make PythonAPI without errors](https://private-user-images.githubusercontent.com/152398875/298506205-10f9c791-d6b0-4c61-ae27-af66953c1518.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYwODUxMTgsIm5iZiI6MTcwNjA4NDgxOCwicGF0aCI6Ii8xNTIzOTg4NzUvMjk4NTA2MjA1LTEwZjljNzkxLWQ2YjAtNGM2MS1hZTI3LWFmNjY5NTNjMTUxOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEyNFQwODI2NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04YWY0MDI3NjAxNGQ4ODMxNjJkMzMxY2U3MGU4YzZhZmQxODE1OGExOTg1ODNkNGE2ZjQ5ODBlOWNiNzQ2Yzg1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.iFsfbcurS_Aa_EiHRw4xekSr_kJkQpzuDkDsx5WoKyo) But in the make launch I got: D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Sensor/SceneCaptureSensor.h(15): fatal error C1083: No se puede abrir el archivo incluir: 'Renderer/Public/GBufferView.h': No such file or directory make: *** [CarlaUE4Editor] Error 6
> 
> ![error make launch](https://private-user-images.githubusercontent.com/152398875/298505913-4cb540db-c85a-4686-80f9-3df658a25506.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYwODUxMTgsIm5iZiI6MTcwNjA4NDgxOCwicGF0aCI6Ii8xNTIzOTg4NzUvMjk4NTA1OTEzLTRjYjU0MGRiLWM4NWEtNDY4Ni04MGY5LTNkZjY1OGEyNTUwNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEyNFQwODI2NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02ZDA1YTczNmM0OWMwNjU4NjE5ZGE0MDYzOTA5MmJlNGU1YjY4Njc4N2M2Mzg1OTQwMjNlZTE2MGI4NDQ5NzZkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.-IhxxW8nT6kqnE2bSdwqZxrK20TcpOPVSOHUeR8hnkk)
> 
> This is the new point I'm stuck at, in case anyone knows how to get out of here.

The problem of D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Sensor/SceneCaptureSensor.h(15): fatal error C1083: Cannot open file include: 'Renderer/Public/GBufferView.h' : No such file or directory
resides in the branch in which it is being built. This way you can fix it:
Clone the repository, make sure you are on the "carla" branch and make sure you rebuild the engine afterwards. Then rebuild Carla version 0.9.15. This should solve the problem.

The GBufferView header cannot be compiled because it is not found in the standard version of Unreal Engine. It is something that has been added to Carla's branch. If you do a git diff between both versions of standard Unreal and Carla's, you'll notice that there are only a few changes, and the addition of GBufferView is one of them.

The solution is reinstalling UE4 and Carla folder from the begining, and that is all.

Avinashnr, 2024-03-20:
> same issue for me. I changed `.\Util\BuildTools\BuildOSM2ODR.bat` line 112 from
>
> ```powershell
>     cmake -G %GENERATOR% %PLATFORM%^
> ```
>
> to
>
> ```powershell
>     cmake -G %GENERATOR% -A x64^
> ```
>
> then [...]

Thanks it worked!

Lyanyv, 2024-09-12:
@Jenson66 Rerun `make PythonAPI`, it should work then.

qshjd-123, 2024-11-27:
I encountered the same problem, how did you solve it?