carla-simulator/carla #6901: Can't installing in Ubuntu 22.04 in 0.9.15 ver.

diaa-woo, 2023-11-13:
CARLA version: 0.9.15
Platform/OS: Ubuntu 22.04
Problem you have experienced: It occurred below errors during building B2 engine in make PythonAPI.

```
Setup.sh: Extracting boost for Python 3.
Building B2 engine..

A C++11 capable compiler is required for building the B2 engine.
Toolset 'clang' does not appear to support C++11.

> clang++ -x c++ -std=c++11  check_cxx11.cpp
check_cxx11.cpp:14:10: fatal error: 'thread' file not found
#include <thread>
         ^~~~~~~~
1 error generated.

** Note, the C++11 capable compiler is _only_ required for building the B2
** engine. The B2 build system allows for using any C++ level and any other
** supported language and resource in your projects.

You can specify the toolset as the argument, i.e.:
    ./build.sh [options] gcc

Toolsets supported by this script are:
    acc, clang, como, gcc, intel-darwin, intel-linux, kcc, kylix, mipspro,
    pathscale, pgi, qcc, sun, sunpro, tru64cxx, vacpp

For any toolset you can override the path to the compiler with the '--cxx'
option. You can also use additional flags for the compiler with the
'--cxxflags' option.

A special toolset; cxx, is available which is used as a fallback when a more
specific toolset is not found and the cxx command is detected. The 'cxx'
toolset will use the '--cxx' and '--cxxflags' options, if present.

Options:
    --help                  Show this help message.
    --verbose               Show messages about what this script is doing.
    --debug                 Build b2 with debug information, and no
                            optimizations.
    --guess-toolset         Print the toolset we can detect for building.
    --cxx=CXX               The compiler exec to use instead of the detected
                            compiler exec.
    --cxxflags=CXXFLAGS     The compiler flags to use in addition to the
                            flags for the detected compiler.

Failed to build B2 build engine
make: *** [Util/BuildTools/Linux.mk:142: setup] Error 1

```

In previous version, I had similar problem but I solved to change `Util/BuildTools/Linux.sh` from this #6381. But this version, I don't have ideas to it can build. I tried to install llvm-10 in Ubuntu 22.04, but it doesn't allow it. 

What should I do?

zhengxh-git, 2023-11-14:
I have same problem
[image]

ABCSchuetze, 2023-11-22:
i have similar problems, but i still get an error even when i install g++-10 and clang-12 

the error i have:
`~/carla/Build/boost-1.80.0-c10-source/tools/build/src/tools/common.jam:1140: in common.find-compiler from module common
error: toolset clang-linux initialization:
error: version '10.0' requested but 'clang++-10.0' not found and version '12.0.1' of default 'clang++' does not match
error: initialized from
`

Does anyone know how to solve the problem?

Ubuntu 22.04
Carla 0.9.15

katcinskiy, 2024-01-25:
@diaa-woo The problem is in clang - it can't find std lib, to fix it, check 

https://stackoverflow.com/questions/54521402/locating-iostream-in-clang-fatal-error-iostream-file-not-found

and delete folder under "Selected GCC installation:" if it does not contain libstdc++.a file. After these steps you will be able to run it.

xavisolesoft, 2024-02-07:
We added additional steps about how to build for Ubuntu22.04 in CARLA build official documentation.
Please, consider following this link from now to build CARLA for Ubuntu 22.04:
[](https://carla.readthedocs.io/en/latest/build_linux/)

xavisolesoft, 2024-02-07:
We added additional steps about how to build for Ubuntu22.04 in CARLA build official documentation.
Please, consider following this link from now to build CARLA for Ubuntu 22.04:
[https://carla.readthedocs.io/en/latest/build_linux/](https://carla.readthedocs.io/en/latest/build_linux/)

Billthekidz, 2024-02-08:
> @diaa-woo Your solution did not help me, here my way to fix it.
> 
> The problem is in clang - it can't find std lib, to fix it, check
> 
> https://stackoverflow.com/questions/54521402/locating-iostream-in-clang-fatal-error-iostream-file-not-found
> 
> and delete folder under "Selected GCC installation:" if it does not contain libstdc++.a file. After these steps you will be able to run it.

Thank you for providing the link. We resolve the issue by finding out the g++/gcc version being used by clang. 
```
clang -v
```
Based on the output, we know that clang is selecting g++ version 12. Then we found out our installation of version 12 does not have `libstdc++.a` in `/usr/lib/gcc/x86_64-linux-gnu/12`. 
We resolved it using `sudo apt install g++-12` that installs `libstdc++.a` to the folder.

hotmedal, 2024-02-16:
> > @diaa-woo Your solution did not help me, here my way to fix it.
> > The problem is in clang - it can't find std lib, to fix it, check
> > https://stackoverflow.com/questions/54521402/locating-iostream-in-clang-fatal-error-iostream-file-not-found
> > and delete folder under "Selected GCC installation:" if it does not contain libstdc++.a file. After these steps you will be able to run it.
> 
> Thank you for providing the link. We resolve the issue by finding out the g++/gcc version being used by clang.
> 
> ```
> clang -v
> ```
> 
> Based on the output, we know that clang is selecting gcc/g++ version 12. Then we found out our installation of version 12 does not have `libstdc++.a` in `/usr/lib/gcc/x86_64-linux-gnu/12`. We resolved it using `sudo apt install g++-12` that installs `libstdc++.a` to the folder.

`sudo apt install g++-12`  should be in the official instructions. It worked for me.

xavisolesoft, 2024-02-16:
> `sudo apt install g++-12` should be in the official instructions. It worked for me.

Thank you for sharing it.

Note that we are already installing g++-7 in the official instructions to maintain the same official toolchain for all Ubuntu versions. As you can see in the next snapshot, g++7 also contains libstdc++.a:
[image]

closed 2024-02-16 by xavisolesoft

JQUP20, 2024-02-22:
> `sudo apt install g++-12` should be in the official instructions. It worked for me.

Good job,it works.

donghaiwang, 2024-09-05:
I have similar problems when i install corresponding g++ version in Ubuntu 24.04.

CaMi1le, 2024-10-21:
> Thank you for sharing it.
>
> Note that we are already installing g++-7 in the official instructions to maintain the same official toolchain for all Ubuntu versions. As you can see in the next [...]

it turns out by 2024/10/21, the Ubuntu 22.04 user following the official guide still have to look back to this issue and use `sudo apt install g++-12` , maybe add a line to tutorial will help, saving 30 mins of time.

RandyWangRaining, 2025-02-06:
> it turns out by 2024/10/21, the Ubuntu 22.04 user following the official guide still have to look back to this issue and use `sudo apt install g++-12` , maybe add a line to tutorial will help, [...]

20.04also helps

Cynthia11281128, 2025-03-01:
> Thank you for providing the link. We resolve the issue by finding out the g++/gcc version being used by clang.
>
> ```
> clang -v
> ```
>
> Based on the output, we know that clang is selecting gcc/g++ version [...]

I an using g++-13, and this also works for me! Thanks!

pdaelm, 2025-03-21:
I had the problem with carla 0.9.15 on a fresh install of ubuntu 20.04.

`
Setup.sh: Extracting boost for Python 3.
Building B2 engine..

A C++11 capable compiler is required for building the B2 engine.
Toolset 'clang' does not appear to support C++11.

> clang++ -x c++ -std=c++11  check_cxx11.cpp
check_cxx11.cpp:14:10: fatal error: 'thread' file not found
#include <thread>
`

Removing g++13 so clang cannot select it, as libstdc++ latest version is 11 on ubuntu 20.04 ...
thx @katcinskiy !

zt282, 2025-05-14:
> it turns out by 2024/10/21, the Ubuntu 22.04 user following the official guide still have to look back to this issue and use `sudo apt install g++-12` , maybe add a line to tutorial will help, [...]

in2025.05.14 we still need that

PabloVD, 2025-05-20:
I also needed to set consistent clang and g++ versions using `update-alternatives`:

```
sudo update-alternatives --install /usr/bin/clang++ clang++ /usr/lib/llvm-10/bin/clang++ 180 &&
sudo update-alternatives --install /usr/bin/clang clang /usr/lib/llvm-10/bin/clang 180 &&
sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-12 180
sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-7 100
```

jiehelin, 2025-08-28:
2025-8-28
clang -12 and gcc instructions are still not included in the documentation. The first few commands may be fatally misleading to users.

[image]