carla-simulator/carla #6902: Semantic segmenation raws labels are not coherent with the documentation

NnaYelsel, 2023-11-13:
_CARLA version_: 0.9.14
_Platform/OS_: Windows
_Problem you have experienced:_
By saving the semantic segmentation label into npy, I didn't get the label value written in the documentation: https://carla.readthedocs.io/en/latest/ref_sensors/#semantic-segmentation-camera but those from https://github.com/carla-simulator/carla/blob/master/LibCarla/source/carla/image/CityScapesPalette.h 

Example: label of the road is 1 and not 7 in my case.

These when, I use a basic function labels_to_cityscapes_palettes in python (ex: https://github.com/carla-simulator/data-collector/blob/master/carla/image_converter.py) the colors are wrong

_Steps to reproduce:_
np_image = np.reshape(np.copy(data.raw_data), (data.height, data.width, 4))
label = np_image[:, :, 2] # class Id is stored in B channel
np.save(npy_file, label)

Then, do we agree that the labels of the raw data are not those of the cityscape palette or did I miss something ?

yyxssm, 2023-12-20:
The same question to me, I think the documentation has not been updated.

I choose to use the label in [https://github.com/carla-simulator/carla/blob/master/LibCarla/source/carla/image/CityScapesPalette.h](url)

qhaas, 2024-01-23:
Noticed this as well in 0.9.15, 11 is sky and 14 is vehicle, which differs from the documentation.

I do like [this comment](https://github.com/carla-simulator/carla/blob/master/LibCarla/source/carla/image/CityScapesPalette.h#L19)
`// Please update documentation if you change this.`

dorn-gerhard, 2024-02-02:
I just experienced the same issue - I just made a pull request PR #7112 (https://github.com/carla-simulator/carla/pull/7112) to update the docs.
Here you can find the new labels including color code:

https://github.com/carla-simulator/carla/blob/d7b45c1e159e6d13296f7a3a4e8b13e6c2d62c18/LibCarla/source/carla/image/CityScapesPalette.h#L35

Hope that helps :)

PR #7112 (2024-02-02): Update semantic labels in doc (changed from version 0.9.13 -> 0.9.14)

#### Description

I updated the semantic labels used since 0.9.14 in the docs

Fixes #6902   

#### Where has this been tested?

  * **Platform(s):** ...
  * **Python version(s):** ...
  * **Unreal Engine version(s):** ...

#### Possible Drawbacks
PR #7122 (2024-02-05): Update semantic labels in doc (changed from version 0.9.13 -> 0.9.14)

Checklist:

  - [ x] Your branch is up-to-date with the `dev` branch and tested with latest changes
  - [ x] Extended the README / documentation, if necessary
  - [ x] Code compiles correctly (did not change any code)
  - [ ] All tests passing with `make check` (only Linux)
  - [ ] If relevant, update CHANGELOG.md with your changes

-->

#### Description

I updated the semantic labels used since 0.9.14 in the docs

Fixes #6902   

#### Where has this been tested?

  * **Platform(s):** ...
  * **Python version(s):** ...
  *  [... cut]
closed 2024-02-02 by Blyron