carla-simulator/carla #6905: Instance Segmentation does not show terrain

yoakim82, 2023-11-13:
CARLA version: 0.9.14 (custom build with included custom objects)
At least Town01 and Town03 affected, probably all maps.

Platform/OS: Ubuntu 22.04 LTS

Problem you have experienced:

When creating snapshots of a scene with both depth, rgb and segmentation sensors, there is an unexpected behavior in the segmentation view of the scene. Note I'm referring to the instance segmentation sensor here.
We use the instance segmentation to capture tight bounding boxes for our objects of interest, but when an object is spawned behind a hill or piece of terrain, the segmentation view will show the object even though it is 100% occluded in the rgb view. This totally ruins the purpose of having semantic segmentation. 

[image]
[image]

What you expected to happen:
I expect the objects being occluded by the terrain to not show up in the segmentation view, which seem to be the case for other occluding objects (e.g. trees/vegetation).
To clarify, the bounding boxes shown in the rgb image are generated only from information in the instance segmentation image.

Steps to reproduce:
Setup a triplet of sensors and navigate to a place in the map like in our enclosed scene.

nikky4D, 2023-12-05:
did you have this solved?

yoakim82, 2023-12-06:
No, I did not. My guess is it's related to how the town is created and how the terrain is labeled semantically.

But I have no clue as to how to fix this. Anyone with deeper UE4 skills is welcome to comment on this.

Yongzhaooo, 2024-04-03:
I met the same situation and it is still not fixed now : (
[image]

AreopagX, 2024-09-11:
related: #5367

tdegraaff, 2025-01-30:
@jorge-kabuto Can now be closed (https://github.com/carla-simulator/carla/pull/8528).

closed 2025-01-31 by Blyron