carla-simulator/carla #6911: LiDAR acquisitions do not register points of actors during replays

FilippoBragato, 2023-11-14:
CARLA version: 0.9.15
Platform/OS: Ubuntu 20.04

[image]

I am trying to gather LiDAR acquisitions from a log generated from the same version of CARLA.
During the generation of the log, the LiDAR sensor works properly. But when I try to get acquisitions during a replay (in SyncMode) the LiDAR is not able to see points related to vehicles or pedestrians.

In the picture on the left, there is the output of the LiDAR, obtained using functions from  `open3d_lidar.py`. On the right, this scene generating the point cloud is shown, the position of the LiDAR is marked with the gray dot above the Seat Leon.

Aleoli2, 2023-12-21:
I had the same issue, the LiDAR didn't show the walkers. I solved it by going back to version 0.9.14. Seeing the changelog, this problem could be related to "Pedestrians with AI or in replayer are now faster around 10x. They have collisions disabled until they hit a vehicle."

Pyrestone, 2024-01-04:
I'm also facing this issue.

More generally, I also found pedestrians to be invisible in lidar even when simulating for the first time as long as an AI walker controller is attached to them. 

Maybe the AI walker controller disables the collision mesh entirely?
Either way, this renders the AIWalkerController pretty much unusable for me.

lfxx, 2024-01-10:
> I'm also facing this issue.
>
> More generally, I also found pedestrians to be invisible in lidar even when simulating for the first time as long as an AI walker controller is attached to them.
>
> Maybe [...]

This is true.AI controller totally destroy lidar's detection on walker.This is a fatal bug which is needed been fixed at once!

sithu31296, 2024-01-25:
I also have the same issue. Please fix it ASAP.

Pyrestone, 2024-03-28:
Did anyone find a workaround for this yet?
@AreopagX posted a workaround for the invisible pedestrians in lidar in #7092, but I'm not sure if/how it's applicable for this replay issue. 
For the moment my strategy has just been to stay with 0.9.14 for the time being.
New carla releases seem to be 80% new features, 5% bug fixes and 15% new bugs in the last two years, so I'm not sure when the next usable version will be...

LuisPoveda, 2024-04-19:
Seems related with #7238 we will be fixing this

imyyf, 2024-05-24:
> I had the same issue, the LiDAR didn't show the walkers. I solved it by going back to version 0.9.14. Seeing the changelog, this problem could be related to "Pedestrians with AI or in replayer are now faster around 10x. They have collisions disabled until they hit a vehicle."

I use carla 0.9.14, but i still face the problem. And issue 7238 seems face the problem using 0.9.14 as well. Do you have any other solutions?

LuisPovedaCano, 2024-05-31:
This issue has been fixed, please update to the latest dev, feel free to reopen the issue if it occurs again.

closed 2024-05-31 by LuisPovedaCano

imyyf, 2024-06-07:
> This issue has been fixed, please update to the latest dev, feel free to reopen the issue if it occurs again.

Will you release a new stable version carla? If yes, when will it be released? Thanks.