carla-simulator/carla #6912: Docker Image: Maps

axelbr, 2023-11-15:
How to use newer maps when running CARLA in Docker? I am using the latest docker image with CARLA version 0.9.15, but I can not find Town 11, 12, 13 and 15.

axelbr, 2023-11-15:
When running ` print(client.get_available_maps())`, I get the following towns:

`['/Game/Carla/Maps/Town04_Opt', '/Game/Carla/Maps/Town10HD', '/Game/Carla/Maps/Town04', '/Game/Carla/Maps/Town01', '/Game/Carla/Maps/Town05', '/Game/Carla/Maps/Town10HD_Opt', '/Game/Carla/Maps/Town02_Opt', '/Game/Carla/Maps/Town03', '/Game/Carla/Maps/Town05_Opt', '/Game/Carla/Maps/Town01_Opt', '/Game/Carla/Maps/Town03_Opt', '/Game/Carla/Maps/Town02']
`

DanielCoelho112, 2023-11-16:
Same issue. It works with CARLA 0.9.14 though.

DeepDuke, 2024-01-23:
I used docker image `0.9.15`, but I found it only have
```python
['/Game/Carla/Maps/Town01', '/Game/Carla/Maps/Town10HD_Opt', '/Game/Carla/Maps/Town04', '/Game/Carla/Maps/Town05_Opt', '/Game/Carla/Maps/Town04_Opt', '/Game/Carla/Maps/Town02_Opt', '/Game/Carla/Maps/Town05', '/Game/Carla/Maps/Town02', '/Game/Carla/Maps/Town10HD', '/Game/Carla/Maps/Town01_Opt', '/Game/Carla/Maps/Town03_Opt', '/Game/Carla/Maps/Town03']

```

siddharthjain38, 2024-03-20:
@DeepDuke @DanielCoelho112 I hope you are able to solve this issue, but if not then you have to include Town 12, 13, 15 in unreal engine before accessing. In unreal engine go to edit -> project setting and search list of maps. Here you have to add map that you wants to use.

axelbr, 2024-03-20:
I ended up doing the following:
- Download the additional maps from github
- Create a new dockerfile based off carla: copy the maps there and run the Import script.

axelbr, 2024-03-20:
```
FROM carlasim/carla:0.9.15
USER root
RUN apt-get update && apt-get install -y wget
RUN wget https://carla-releases.s3.eu-west-3.amazonaws.com/Linux/AdditionalMaps_0.9.15.tar.gz
RUN mv AdditionalMaps_0.9.15.tar.gz /home/carla/Import
WORKDIR /home/carla/Import
RUN tar -xf AdditionalMaps_0.9.15.tar.gz
WORKDIR /home/carla
RUN ./ImportAssets.sh
RUN rm -f /home/carla/Import/AdditionalMaps_0.9.15.tar.gz
USER carla
```

closed 2024-03-20 by axelbr