carla-simulator/carla #6913: Digital Twin tool crashing the engine

KPicherit, 2023-11-15:
Hello everyone,
I saw with great enthusiasm that CARLA0.9.15 comes with procedural environment generation based on osm data.
I tried to follow the tutorial but the Engine crashes during environment generation.
The XODR file is generated, as well as the tiles (that become available after the crash once engine is restarted), but nothing is present in the tiles when I run manual_control.py
Am I missing something here ?

Note: I run the Widget from editor mode, which I guess is the right way to go.

---------------

CARLA version: 0.9.15
Platform/OS: Ubuntu 22.04, 24GB Ram, 
Problem you have experienced: Crash following [digital twin tutorial](https://carla.readthedocs.io/en/latest/adv_digital_twin/)
What you expected to happen: Get a procedurally generated map based on selected OpenStreetmap area
Steps to reproduce: Follow the tutorial trying to import a [test road](https://www.openstreetmap.org/export#map=15/43.7285/3.3286) ([overpassAPI link](https://overpass-api.de/api/map?bbox=3.3192,43.7197,3.3384,43.7354))

----------------
Step 1

[image]

----------------
Step 2

Tried this with several OpenStreetmap locations in different countries. Here: ([overpassAPI link](https://overpass-api.de/api/map?bbox=3.3192,43.7197,3.3384,43.7354))

[image]
Click Generate

----------------
Crash

[image]

```
FileName TestMap, AssetName TestMap .
[None] PostInit
=== Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to read memory at address 0x0000000000000000

Fatal error!

0x00007f2f2403b748 libUE4Editor-Engine.so!UMaterialInterface::AnalyzeMaterialProperty(EMaterialProperty, int&, bool&) [/data/UE_4.26_carla/Engine/Source/Runtime/Engine/Private/Materials/MaterialShared.cpp:3435]
0x00007f2e400e93fb libUE4Editor-MeshMergeUtilities.so!FMeshMergeUtilities::DetermineMaterialVertexDataUsage(TArray<bool, TSizedDefaultAllocator<32> >&, TArray<UMaterialInterface*, TSizedDefaultAllocator<32> > const&, UMaterialOptions const*) const [/data/UE_4.26_carla/Engine/Source/Developer/MeshMergeUtilities/Private/MeshMergeUtilities.cpp:519]
0x00007f2e400f5b04 libUE4Editor-MeshMergeUtilities.so!FMeshMergeUtilities::MergeComponentsToStaticMesh(TArray<UPrimitiveComponent*, TSizedDefaultAllocator<32> > const&, UWorld*, FMeshMergingSettings const&, UMaterialInterface*, UPackage*, FString const&, TArray<UObject*, TSizedDefaultAllocator<32> >&, FVector&, float, bool) const [/data/UE_4.26_carla/Engine/Source/Developer/MeshMergeUtilities/Private/MeshMergeUtilities.cpp:2079]
0x00007f2dceba5d81 libUE4Editor-EditorScriptingUtilities.so!UEditorLevelLibrary::MergeStaticMeshActors(TArray<AStaticMeshActor*, TSizedDefaultAllocator<32> > const&, FEditorScriptingMergeStaticMeshActorsOptions const&, AStaticMeshActor*&) [/data/UE_4.26_carla/Engine/Plugins/Editor/EditorScriptingUtilities/Source/EditorScriptingUtilities/Private/EditorLevelLibrary.cpp:1338]
```

------------- 

Here is the [full crash report](https://github.com/carla-simulator/carla/files/13364547/crashinfo-CarlaUE4-pid-53445-37F3D5068F85497DA3557AC2CBCB7713.zip)

Other information (documentation you consulted, workarounds you tried):
* Unreal and Carla fresh build
* Tried both `Using a URL` and `Downloading an OSM file and navigating in the interface` from tutorial
* Seems there is some problems with Walker animations not found as well, I don't know if this is related somehow. I updated my content folder and am now at 20231108_c5101a5 which corresponds to version defined in `Util/ContentVersions.txt`
*  Tried filtering the osm file to only keep the roads using `osmfilter`
* Tried to generate really small zones

Thanks for your help !

domajstorovic, 2023-11-16:
I can also confirm the same issue.

My setup is pretty much the same:

- Ubuntu 22.04 (64G RAM, RTX4090)
- 0.9.15 built from source, running in editor mode

omerfarukadastec, 2023-11-17:
I can also confirm too.

My setup is nearly same too.

- Ubuntu 20.04 (128GB RAM, RTX3090)
- 0.9.15 CARLA fresh build, running in editor mode

xX-Conan-Xx, 2023-11-27:
The same thing happened to me.

I was running CARLA UE4 Editor inside a docker container from this [tutorial](https://carla.readthedocs.io/en/0.9.15/build_docker_unreal/)
[image]
When I press generate map, after 30 secs. This crash info pops up.
[image]

```
LoginId:-000003e8
EpicAccountId:

Caught signal 11 Segmentation fault

libUE4Editor-Engine.so!UMaterialInterface::AnalyzeMaterialProperty(EMaterialProperty, int&, bool&) [/home/carla/UE4.26/Engine/Source/Runtime/Engine/Private/Materials/MaterialShared.cpp:3435]
libUE4Editor-MeshMergeUtilities.so!FMeshMergeUtilities::DetermineMaterialVertexDataUsage(TArray<bool, TSizedDefaultAllocator<32> >&, TArray<UMaterialInterface*, TSizedDefaultAllocator<32> > const&, UMaterialOptions const*) const [/home/carla/UE4.26/Engine/Source/Developer/MeshMergeUtilities/Private/MeshMergeUtilities.cpp:519]
libUE4Editor-MeshMergeUtilities.so!FMeshMergeUtilities::MergeComponentsToStaticMesh(TArray<UPrimitiveComponent*, TSizedDefaultAllocator<32> > const&, UWorld*, FMeshMergingSettings const&, UMaterialInterface*, UPackage*, FString const&, TArray<UObject*, TSizedDefaultAllocator<32> >&, FVector&, float, bool) const [/home/carla/UE4.26/Engine/Source/Developer/MeshMergeUtilities/Private/MeshMergeUtilities.cpp:2079]
libUE4Editor-EditorScriptingUtilities.so!UEditorLevelLibrary::MergeStaticMeshActors(TArray<AStaticMeshActor*, TSizedDefaultAllocator<32> > const&, FEditorScriptingMergeStaticMeshActorsOptions const&, AStaticMeshActor*&) [/home/carla/UE4.26/Engine/Plugins/Editor/EditorScriptingUtilities/Source/EditorScriptingUtilities/Private/EditorLevelLibrary.cpp:1338]
libUE4Editor-EditorScriptingUtilities.so!UEditorLevelLibrary::execMergeStaticMeshActors(UObject*, FFrame&, void*) [/home/carla/UE4.26/Engine/Plugins/Editor/EditorScriptingUtilities/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/EditorScriptingUtilities/EditorLevelLibrary.gen.cpp:419]
libUE4Editor-CoreUObject.so!UObject::execCallMathFunction(UObject*, FFrame&, void*) [/home/carla/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:930]
libUE4Editor-CoreUObject.so!UObject::execLetBool(UObject*, FFrame&, void*) [/home/carla/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:2771]
libUE4Editor-CoreUObject.so!ProcessLocalScriptFunction(UObject*, FFrame&, void*) [/home/carla/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1094]
libUE4Editor-CoreUObject.so!void ProcessScriptFunction<void (*)(UObject*, FFrame&, void*)>(UObject*, UFunction*, FFrame&, void*, void (*)(UObject*, FFrame&, void*)) [/home/carla/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:895]
libUE4Editor-CoreUObject.so!ProcessLocalFunction(UObject*, UFunction*, FFrame&, void*) [/home/carla/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1128]
libUE4Editor-CoreUObject.so!UObject::ProcessContextOpcode(FFrame&, void*, bool) [/home/carla/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:2867]
libUE4Editor-CoreUObject.so!ProcessLocalScriptFunction(UObject*, FFrame&, void*) [/home/carla/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1094]
libUE4Editor-CoreUObject.so!void ProcessScriptFunction<void (*)(UObject*, FFrame&, void*)>(UObject*, UFunction*, FFrame&, void*, void (*)(UObject*, FFrame&, void*)) [/home/carla/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:895]
libUE4Editor-CoreUObject.so!ProcessLocalFunction(UObject*, UFunction*, FFrame&, void*) [/home/carla/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1128]
libUE4Editor-CoreUObject.so!ProcessLocalScriptFunction(UObject*, FFrame&, void*) [/home/carla/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1094]
libUE4Editor-CoreUObject.so!void ProcessScriptFunction<void (*)(UObject*, FFrame&, void*)>(UObject*, UFunction*, FFrame&, void*, void (*)(UObject*, FFrame&, void*)) [/home/carla/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:895]
libUE4Editor-CoreUObject.so!ProcessLocalFunction(UObject*, UFunction*, FFrame&, void*) [/home/carla/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1128]
libUE4Editor-CoreUObject.so!ProcessLocalScriptFunction(UObject*, FFrame&, void*) [/home/carla/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1094]
libUE4Editor-CoreUObject.so!UFunction::Invoke(UObject*, FFrame&, void*) [/home/carla/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/Class.cpp:5588]
libUE4Editor-CoreUObject.so!UObject::ProcessEvent(UFunction*, void*) [/home/carla/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1990]
libUE4Editor-CarlaTools.so!UOpenDriveToMap::GenerationFinished(FVector, FVector) [/home/carla/carla/Unreal/CarlaUE4/Plugins/CarlaTools/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/CarlaTools/OpenDriveToMap.gen.cpp:204]
libUE4Editor-CarlaTools.so!UOpenDriveToMap::GenerateTile() [/home/carla/carla/Unreal/CarlaUE4/Plugins/CarlaTools/Source/CarlaTools/Private/OpenDriveToMap.cpp:336]
libUE4Editor-CarlaTools.so!UGenerateTileCommandlet::Main(FString const&) [/home/carla/carla/Unreal/CarlaUE4/Plugins/CarlaTools/Source/CarlaTools/Private/Commandlet/GenerateTileCommandlet.cpp:80]
UE4Editor!FEngineLoop::PreInitPostStartupScreen(char16_t const*) [/home/carla/UE4.26/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:3369]
[... cut ...]
When I pressed close and sending again the UE4 editor kept on running again and I found a StreetMapActor in the world outlier
[image]
And after I click on BuildMesh inside the details of that actor a white mesh of the buildings appear
[image]
Which is actually correct, but the road infomation is not shown and the buildings are not textured as well.

I am a bit lost here. Seems like some function worked but others didn't.
Here is the full output from my terminal:
[TerminalLog1.txt](https://github.com/carla-simulator/carla/files/13469987/TerminalLog1.txt)

gitkatrin, 2023-12-12:
Same problem for me, if there are houses on the map. It builds the "Open Drive" but not the "Static" folder inside the map folder and you cannot see anything. The tip from  @xX-Conan-Xx with "BildMesh" works also for me but still no streets and trees and and...

Somehow it works with maps without houses on. Streets and trees are no problem. As you can see here is also the "Static" folder.
[image]

zjr-bit, 2023-12-25:
I meet the same situation as @gitkatrin : when the map includes buildings, it will crash.

scretvaule, 2023-12-28:
Why is there no plugin file after I compile
[image]

matejm42, 2024-01-08:
> Why is there no plugin file after I compile [...]
@scretvaule try #7053 I might have the same problem

And to others, thank you for your tips. I was able to duplicate your steps. With buildings, it always crashes, but I can generate nice mash, if I let the simulator running, but no streets. The tool works, if there are no buildings, although the generated map is not really similar to the original:
Generated map:
[image]
Original
[image]
And it has o lot of buggy trees on the road etc.
[image]
[image]

Sivagopinathreddy-Vinta, 2024-01-28:
I can also confirm too.

My setup is nearly same too.

    Ubuntu 18.04 (32GB RAM, RTX4070)
    0.9.15 CARLA fresh build, running in editor mode

> Hello everyone, I saw with great enthusiasm that CARLA 0.9.15 comes with procedural environment generation based on osm data. I tried to follow the tutorial but the Engine crashes during environment generation. The XODR file is generated, as well as the tiles (that become available after the crash once engine is restarted), but nothing is present in the tiles when I run manual_control.py Am I missing something here ?
> 
> Note: I run the Widget from editor mode, which I guess is the right way to go.
> 
> CARLA version: 0.9.15 Platform/OS: Ubuntu 22.04, 24GB Ram, Problem you have experienced: Crash following [digital twin tutorial](https://carla.readthedocs.io/en/latest/adv_digital_twin/) What you expected to happen: Get a procedurally generated map based on selected OpenStreetmap area Steps to reproduce: Follow the tutorial trying to import a [test road](https://www.openstreetmap.org/export#map=15/43.7285/3.3286) ([overpassAPI link](https://overpass-api.de/api/map?bbox=3.3192,43.7197,3.3384,43.7354))
> 
> Step 1
> <img alt="image" width="600" height="400" src="https://private-user-images.githubusercontent.com/107474213/283095766-c7c6c9d6-d73e-46d1-b83d-4874c8b16d05.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY0Mzk4MzMsIm5iZiI6MTcwNjQzOTUzMywicGF0aCI6Ii8xMDc0NzQyMTMvMjgzMDk1NzY2LWM3YzZjOWQ2LWQ3M2UtNDZkMS1iODNkLTQ4NzRjOGIxNmQwNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEyOFQxMDU4NTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05MzJhZWZhZGY4NDk3YmNkMTMwN2MzZTk0MTY1MThlMGI5NGI2YTJhYWQzYTUwZjU4NzZiZTY0NWY0MzhmZmIyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.D_FdHQxp2ZjeQGdwLX7MKQtLK1wUufK-1ueB-kU6c6Y">
> 
> Step 2
> 
> Tried this with several OpenStreetmap locations in different countries. Here: ([overpassAPI link](https://overpass-api.de/api/map?bbox=3.3192,43.7197,3.3384,43.7354))
> <img alt="image" width="600" height="400" src="https://private-user-images.githubusercontent.com/107474213/283111529-26701c5d-453a-4b1d-8fd4-5decdfa62eab.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY0Mzk4MzMsIm5iZiI6MTcwNjQzOTUzMywicGF0aCI6Ii8xMDc0NzQyMTMvMjgzMTExNTI5LTI2NzAxYzVkLTQ1M2EtNGIxZC04ZmQ0LTVkZWNkZmE2MmVhYi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEyOFQxMDU4NTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02NjQ5Mzk0ZTUyNzczMTlmY2Q0MmRlZDFlNWM0Mzk5MTRhOWIzNGYyNDJhZWE3ZWQxYjMyOTMxMzkyZWIzY2Y5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.kfmxny9lCuZug34Y5OrXokTV1FBbXDE2fYLQ-wyReDw"> Click Generate
> 
> Crash
> <img alt="image" width="600" height="400" src="https://private-user-images.githubusercontent.com/107474213/283112144-f741f4ce-f4eb-4e1a-85ef-f526eca572a9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY0Mzk4MzMsIm5iZiI6MTcwNjQzOTUzMywicGF0aCI6Ii8xMDc0NzQyMTMvMjgzMTEyMTQ0LWY3NDFmNGNlLWY0ZWItNGUxYS04NWVmLWY1MjZlY2E1NzJhOS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEyOFQxMDU4NTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04MGQwN2IxYzhhZjhkOGRjODBiMDU5MTBhNWM5YjgzZmFiNTMwZGQ3Mjg3ZDhkNGQxNTUwNzk0N2I2NzM0ZWVmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.3Oi3jt-GGAUCpvPdJGXjjbIw4t1RJwemx-nEbDIMmds">
> 
> ```
> FileName TestMap, AssetName TestMap .
> [None] PostInit
> === Critical error: ===
> Unhandled Exception: SIGSEGV: invalid attempt to read memory at address 0x0000000000000000
> 
> Fatal error!
> 
> 0x00007f2f2403b748 libUE4Editor-Engine.so!UMaterialInterface::AnalyzeMaterialProperty(EMaterialProperty, int&, bool&) [/data/UE_4.26_carla/Engine/Source/Runtime/Engine/Private/Materials/MaterialShared.cpp:3435]
> 0x00007f2e400e93fb libUE4Editor-MeshMergeUtilities.so!FMeshMergeUtilities::DetermineMaterialVertexDataUsage(TArray<bool, TSizedDefaultAllocator<32> >&, TArray<UMaterialInterface*, TSizedDefaultAllocator<32> > const&, UMaterialOptions const*) const [/data/UE_4.26_carla/Engine/Source/Developer/MeshMergeUtilities/Private/MeshMergeUtilities.cpp:519]
> 0x00007f2e400f5b04 libUE4Editor-MeshMergeUtilities.so!FMeshMergeUtilities::MergeComponentsToStaticMesh(TArray<UPrimitiveComponent*, TSizedDefaultAllocator<32> > const&, UWorld*, FMeshMergingSettings const&, UMaterialInterface*, UPackage*, FString const&, TArray<UObject*, TSizedDefaultAllocator<32> >&, FVector&, float, bool) const [/data/UE_4.26_carla/Engine/Source/Developer/MeshMergeUtilities/Private/MeshMergeUtilities.cpp:2079]
> 0x00007f2dceba5d81 libUE4Editor-EditorScriptingUtilities.so!UEditorLevelLibrary::MergeStaticMeshActors(TArray<AStaticMeshActor*, TSizedDefaultAllocator<32> > const&, FEditorScriptingMergeStaticMeshActorsOptions const&, AStaticMeshActor*&) [/data/UE_4.26_carla/Engine/Plugins/Editor/EditorScr [... cut ...]
>     * Seems there is some problems with Walker animations not found as well, I don't know if this is related somehow. I updated my content folder and am now at 20231108_c5101a5 which corresponds to version defined in `Util/ContentVersions.txt`
> 
>     * Tried filtering the osm file to only keep the roads using `osmfilter`
> 
>     * Tried to generate really small zones
> 
> 
> Thanks for your help !

I can also confirm too.

My setup is nearly same too.

    Ubuntu 18.04 (32GB RAM, RTX4070)
    0.9.15 CARLA fresh build, running in editor mode

andiven, 2024-02-02:
I can also confirm too.

```
Ubuntu 22.04.3  (24GB, RTX 3090)
0.9.15  and master (2024.1.20)
```

myersryan, 2024-03-06:
Same issue,

```
Ubuntu 20.04  (64GB, RTX 4090)
0.9.15  and master
```

donghaiwang, 2024-03-07:
Same issue,
`Windows 10 (64G, RTX 3090) 
test in 0.9.15 and dev branch`

Blyron, 2024-05-17:
Material crash is fixed in dev branch. For windows we are not gonna be able to generate largemaps yet.

donghaiwang, 2024-05-20:
> Material crash is fixed in dev branch. For windows we are not gonna be able to generate largemaps yet.

Thank you for fixing this issue, but do you have any plans or ideas for fixing the issues that only the first tile but other map tiles have not been generated in windows.
[image]
[image]

Blyron, 2024-05-21:
It should be fixed and working as expected now.
I will be closing this issue, if something comes up, please feel free to reopen

closed 2024-05-21 by Blyron

diespondency, 2024-08-01:
I still have the same issue as described by others, the generation of the map sometimes works, sometimes it doesn't. When it works, only the buildings are generated and all the information about the roads is gone. It's not clear to me why for some osm's it works and for others everything crashes

joeyleehk, 2024-09-27:
I have tried to go through the whole digital twin tools in Carla 0.9.15 version and I also faced many compile and link issues when built the osmrenderer and CarlaToolsContent/OnroadMapGenerator. However, I figure them out. Here is a list of corresponding solutions: 
1. make osmrenderer: This one needs dependencies and built libosmscout and lunasvg packages. The configuration file locates in ${CARLA_BUILD_FOLDER}/Util/BuildTools/BuildOSMRenderer.sh. In this .sh file, you can see the libosmscout with specific version on Github can be cloned. 

2. But for lunasvg, it will clone the lastest version from Github. 

- This one will cause the error "`error: ‘class lunasvg::Bitmap’ has no member named ‘valid`’". The reason is the lastest lunasvg has been removed this function from its source code, replacing with isNull(). To figure this issue, it is better to manually colne the [v.2.3.8](https://github.com/sammycage/lunasvg/releases/tag/v2.3.8) to the ${CARLA_BUILD_FOLDER}/lunasvg-source (create the folder by yourself). And the comment these lines in .sh file: 
> #echo "Cloning luna-svg"
> #if [ ! -d ${LUNASVG_SOURCE_FOLDER} ] ; then
> #git clone ${LUNASVG_REPO} ${LUNASVG_SOURCE_FOLDER}
> #fi

After this lunasvg can be built smoothly. 
For Authors: would you please add the specific lunasvg COMMIT code in this .sh file? This can avoid the has no member issue. 

3. When built osm-map-renderer, it is suggested to run make PythonAPI firstly. Because the boost-1.80.0-c10-install is needed for building osm-map-renderer. Also, please modify the 

```
cmake -DCMAKE_CXX_FLAGS="-std=c++17 -g -pthread -I${CARLA_BUILD_FOLDER}/boost-1.80.0-c10-install/include" \
    ${OSM_RENDERER_SOURCE}
```
to 
 ```
cmake -DCMAKE_CXX_FLAGS="-std=c++17 -g -pthread -I${CARLA_BUILD_FOLDER}/boost-1.80.0-c10-install/include" \
-DCMAKE_EXE_LINKER_FLAGS="-L${CARLA_BUILD_FOLDER}/boost-1.80.0-c10-install/lib -lboost_system"\
    ${OSM_RENDERER_SOURCE}
```
to link the .a .so file correspondingly. 
4. The issue "usr/include/boost/system/error_code.hpp:206: undefined reference to boost::system::generic_category()'" could be raised during osmrenderer buit. Please add this command in ${OSM_RENDERER_SOURCE}/CMakeLists.txt: add_definitions(-DBOOST_ERROR_CODE_HEADER_ONLY).
5. When osmrenderer is correctly built, do make PythonAPI. 
6. After launch CarlaUE4, you can use CarlaToolsContent/OnroadMapGenerator to generate the whole scene, including buildings. PLEASE MAKE SURE THERE NO SPECIFIC SYMBOL LIKE &|* IN YOUR DOWNLOADED OSM FILE. If these symbols exist please remove or modify them in the OSM file. 
7. After this, run ./build.sh <osm_dir> in {CARLA_ROOT}/Build/libosmcout-source/maps/ to cook the osm file. Then, open UW_OnRoadMainWidget in Unreal and copy the file dir into Select OSM Database, and Please DO NOT CLICK any Button, otherwise the Unreal will crash. 
8. Open a new cmd under {CARLA_ROOT}/Build/osm-world-renderer-build and run ./osm-world-renderer. Until you see "Starting renderer C
┌ Waiting Command.."
9. Click Config and Render in UW_OnRoadMainWidget, also you need to input the geoloaction in Navigation Settings, which can be checked in OSM webpage. Then, you can see the osm map rendered.
10. IMPORTANT: after all setting, please input you File Name, and input you downloaded and modified local .osm file in XODR OR OSM FilePath. Then, click the Generate and have a cup of coffee. 
11. Enjoy the scene you created. Here is my example: 
[image]

```

rohanNkhaire, 2024-10-10:
@joeyleehk Hey, I followed your instructions. However, I am getting invalid character error in the building name. I modified the downloaded .osm file - the original name tag had v="xxxx &amp; xxxx", I modified it to just v="Building". Then built the. Osm using libosmscout. Then I tried to generate the map and I still get the error invalid character xxxx_&amp;_xxxx

joeyleehk, 2024-10-16:
Hi Rohan, have you tried to input the dir of your modified .osm file in the XDOR OR OSM FilePath and define the Geo Coords Orgin. otherwise, it will download the .osm from the OpenStreetMap website.

zihaosheng, 2025-01-31:
Hi @joeyleehk , thanks for your sharing! I followed your steps and succeeded in the first 9 steps. But I failed in Step 10. 
I don't quite understand what you mean by `input you downloaded and modified local .osm file in XODR OR OSM FilePath`, and I didn’t see where I can input the XODR OR OSM FilePath. 
Could you please provide more details on how this is done? Thank you so much!

IrfanAli91170, 2025-02-19:
Excuse me, i just install the Digital Twins rool and all built well. 
But, following the instructions to render OSM maps in CARLA, the rendered map does not appear in the UW_OnRoadMainWidget after Step 9 (Config and Render). 
No errors are displayed, but the widget remains blank. 

These are the steps that I followed :

I first compiled 'make osmrenderer' without any issues. Then, I cooked the OSM file, which I extracted from the OpenStreetMap website for a small area, using ./build.sh, and it worked fine.

Next, I launched CARLA with 'make launch', then opened the UW_OnRoadMainWidget widget. I correctly specified the path to the cooked OSM folder (without clicking any buttons).

In a second terminal, I ran ./osm-world-renderer, and I could see the message:

Starting renderer C
┌ Waiting Command..

Finally, I clicked on 'Config' and 'Render' in the widget, but nothing happened.

I urgently need help with this issue for my work. Could you please assist me?

IrfanAli91170, 2025-02-20:
That's the message when the UE4 crashed :

[image]

[CarlaUE4.log](https://github.com/user-attachments/files/18884091/CarlaUE4.log)

LoginId:7336a0be5b4d913d4c77d8506790bfe7-000003e8
EpicAccountId:

Assertion failed: (Index >= 0) & (Index < ArrayNum) [File:Runtime/Core/Public/Containers/Array.h] [Line: 674] Array index out of bounds: 1 from an array of size 1

libUE4Editor-Core.so!FGenericPlatformMisc::RaiseException(unsigned int) [/home/scarlet/UE4.26/Engine/Source/Runtime/Core/Private/GenericPlatform/GenericPlatformMisc.cpp:472]
libUE4Editor-Core.so!FOutputDevice::LogfImpl(char16_t const*, ...) [/home/scarlet/UE4.26/Engine/Source/Runtime/Core/Private/Misc/OutputDevice.cpp:61]
libUE4Editor-Core.so!AssertFailedImplV(char const*, char const*, int, char16_t const*, __va_list_tag*) [/home/scarlet/UE4.26/Engine/Source/Runtime/Core/Private/Misc/AssertionMacros.cpp:104]
libUE4Editor-Core.so!FDebug::CheckVerifyFailedImpl(char const*, char const*, int, char16_t const*, ...) [/home/scarlet/UE4.26/Engine/Source/Runtime/Core/Private/Misc/AssertionMacros.cpp:458]
libUE4Editor-CarlaTools.so!UMapPreviewUserWidget::UpdateLatLonCoordProperties() [/home/scarlet/carla/Unreal/CarlaUE4/Plugins/CarlaTools/Source/CarlaTools/Private/MapPreviewUserWidget.cpp:205]
libUE4Editor-CoreUObject.so!UFunction::Invoke(UObject*, FFrame&, void*) [/home/scarlet/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/Class.cpp:5588]
libUE4Editor-CoreUObject.so!UObject::CallFunction(FFrame&, void*, UFunction*) [/home/scarlet/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:997]
libUE4Editor-CoreUObject.so!UObject::ProcessContextOpcode(FFrame&, void*, bool) [/home/scarlet/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:2867]
libUE4Editor-CoreUObject.so!ProcessLocalScriptFunction(UObject*, FFrame&, void*) [/home/scarlet/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1094]
libUE4Editor-CoreUObject.so!void ProcessScriptFunction<void (*)(UObject*, FFrame&, void*)>(UObject*, UFunction*, FFrame&, void*, void (*)(UObject*, FFrame&, void*)) [/home/scarlet/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:895]
libUE4Editor-CoreUObject.so!ProcessLocalFunction(UObject*, UFunction*, FFrame&, void*) [/home/scarlet/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1128]
libUE4Editor-CoreUObject.so!ProcessLocalScriptFunction(UObject*, FFrame&, void*) [/home/scarlet/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1094]
libUE4Editor-CoreUObject.so!void ProcessScriptFunction<void (*)(UObject*, FFrame&, void*)>(UObject*, UFunction*, FFrame&, void*, void (*)(UObject*, FFrame&, void*)) [/home/scarlet/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:895]
libUE4Editor-CoreUObject.so!ProcessLocalFunction(UObject*, UFunction*, FFrame&, void*) [/home/scarlet/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1128]
libUE4Editor-CoreUObject.so!ProcessLocalScriptFunction(UObject*, FFrame&, void*) [/home/scarlet/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1094]
libUE4Editor-CoreUObject.so!UFunction::Invoke(UObject*, FFrame&, void*) [/home/scarlet/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/Class.cpp:5588]
libUE4Editor-CoreUObject.so!UObject::ProcessEvent(UFunction*, void*) [/home/scarlet/UE4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1990]
libUE4Editor-UMG.so!void TScriptDelegate<FWeakObjectPtr>::ProcessDelegate<UObject>(void*) const [/home/scarlet/UE4.26/Engine/Source/Runtime/Core/Public/UObject/ScriptDelegates.h:247]
libUE4Editor-UMG.so!void TMulticastScriptDelegate<FWeakObjectPtr>::ProcessMulticastDelegate<UObject>(void*) const [/home/scarlet/UE4.26/Engine/Source/Runtime/Core/Public/UObject/ScriptDelegates.h:493]
libUE4Editor-UMG.so!UButton::SlateHandleClicked() [/home/scarlet/UE4.26/Engine/Source/Runtime/UMG/Private/Components/Button.cpp:203]
libUE4Editor-UMG.so!TBaseUObjectMethodDelegateInstance<false, UButton, FReply (), FDefaultDelegateUserPolicy>::Execute() const [/home/scarlet/UE4.26/Engine/Source/Runtime/Core/Public/Delegates/DelegateInstancesImpl.h:593]
libUE4Editor-Slate.so!SButton::ExecuteOnClick() [/home/scarlet/UE4.26/Engine/Source/Runtime/Slate/Private/Widgets/Input/SButton.cpp:385]
libUE4Editor-Slate.so!SButton::OnMouseButtonUp(FGeometry const&, FPointerEvent const&) [/home/scarlet/UE4.26/Engine/Source/Runtime/Slate/Private/Widgets/Input/SButton.cpp:304]
libUE4Editor-Slate.so!FSlateApplication::RoutePointerUpEvent(FWidgetPath const&, FPointerEvent const&) [/home/scarlet/UE4.26/Engine/Source/Runtime/Slate/Private/Framework/Application/SlateApplication.cpp:4823]
libUE4Editor-Slate.so!FSlateApplication::ProcessMouseButtonUpEvent(FPointerEvent const&) [/home/scarlet/UE4.26/Engine/Source/Runtime/Slate/Private/Framework/Application/SlateApplication.cpp:5363]
libUE4Editor-Slate.so!FSlateApplication::OnMouseUp(EMouseButtons::Type, FVector2D) [/home/scarlet/UE4.26/Engine/Source/Runtime/Slate/Private/Framework/Application/SlateApplication.cpp:5328]
libUE4Editor-ApplicationCore.so!FLinuxApplication::ProcessDeferredMessage(SDL_Event) [/home/scarlet/UE4.26/Engine/Source/Runtime/ApplicationCore/Private/Linux/LinuxApplication.cpp:370]
libUE4Editor-ApplicationCore.so!FLinuxPlatformApplicationMisc::PumpMessages(bool) [/home/scarlet/UE4.26/Engine/Source/Runtime/ApplicationCore/Private/Linux/LinuxPlatformApplicationMisc.cpp:428]
UE4Editor!FEngineLoop::Tick() [/home/scarlet/UE4.26/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:4748]
UE4Editor!GuardedMain(char16_t const*) [/home/scarlet/UE4.26/Engine/Source/Runtime/Launch/Private/Launch.cpp:171]
libUE4Editor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)()) [/home/scarlet/UE4.26/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:264]
libc.so.6!__libc_start_main(+0xe6)

ponljung, 2025-04-13:
I tried to follow above steps from @joeyleehk but seem to still have similar issues as the @KPicherit had in the beginning. I did also go over the OSM file to check for mentioned characters but the issue is still present, i.e it crashes during generation. 

Steps: 

[image]

Selecting the area: 

[image]

I see during generation that this error log is created: 

**[2025.04.13-15.40.26:152][165]LogCarlaTools: Error: Creating New Object**

The generation is then terminated and a log dump from UE is created. Closing without sending still keeps the generation going in the background.

[image]

Running Carla version 0.9.15.2

Full log can be found below:
[log.txt](https://github.com/user-attachments/files/19726083/log.txt)