carla-simulator/carla #6917: CARLA Hangs during Execution: Seeking Insights on GDB Thread Analysis

xiaobanni, 2023-11-16:
CARLA version: 0.9.10
Platform/OS: 18.04.1-Ubuntu
Problem you have experienced: CARLA hangs during execution
What you expected to happen: maybe there's a deadlock
Steps to reproduce: maybe it is triggered under certain conditions, so I've attached a full stack analysis
Other information (documentation you consulted, workarounds you tried):

I am encountering an issue where my python(pytorch) program, which interacts with the CARLA simulator, hangs during execution. To investigate this, I have attached a gdb file with thread analysis (`gdb.txt`) captured at the point where the program becomes unresponsive.

**Issue Description:**
- The program involves multiple threads and seems to be stuck in a waiting state.
- Several threads are in a wait state, particularly with calls to `futex_wait` and PyTorch's autograd engine functions.
- One thread appears to be waiting on a CUDA-related operation (`libcuda.so`), suggesting a potential wait for GPU resources or a GPU operation to complete.
- The main thread is waiting in a CARLA function, specifically in `carla::traffic_manager::TrafficManagerLocal::SynchronousTick`.

Problem figures for references:
[image]
[image]
[image]
[image]

I am attaching the [gdb.txt](https://github.com/carla-simulator/carla/files/13377903/gdb.txt) for reference and would greatly appreciate any insights or suggestions on how to diagnose and resolve this issue. The gdb output includes the state of all threads at the time of the hang.

Thank you for your assistance!

SasukeBLX, 2023-11-17:
I just looked for similar issues elsewhere. I found it at below site.
https://github.com/microsoft/onnxruntime/issues/13993
That issue seemed to be solved by changing LD_LIBRARY_PATH.
Kindly check it.

xiaobanni, 2023-11-18:
@SasukeBLX LD_LIBRARY_PATH is a broad concept and the issues mentioned in the provided link do not seem to closely resemble the situation I'm encountering with CARLA. Therefore, I'm unable to determine a further solution at this point. Thank you for your attention and I welcome any additional suggestions.

xiaobanni, 2023-11-20:
I get a new thread analysis file [gdb.txt](https://github.com/carla-simulator/carla/files/13407954/gdb.txt) when I encounter the question again. I think there are some common issues that may be helpful to solve the problem. It seems the main thread using the Carla server is stuck around the following functions:

* `carla::traffic_manager::TrafficManager::Tick`
* `carla::client::detail::Simulator::Tick`
* `carla::client::World::Tick`

I hope this file will be helpful in addressing these issues.