carla-simulator/carla #6921: Error in make pythonAPI using VS 2022

Nanakea, 2023-11-17:
I have done changing the generator versions in most of the library and all of the dependency could be compiled now. But osm2odr-visualstudio just can't be compiled (I have done using VS 2019 before and it worked just fine)

Because OSM2ODR build is failing, dist folder can't be create so I couldn't use make launch command. Here's the error log:

```
C:\carla>make pythonAPI GENERATOR="Visual Studio 17 2022" -d
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
これはフリーソフトウェアです. 利用許諾についてはソースを
ご覧ください.
商業性や特定の目的への適合性の如何に関わらず, 無保証です.

This program built for i386-pc-mingw32
find_and_set_shell パス探索で default_shell = C:/Program Files/Git/bin/sh.exe にセットしました
makefile を読み込みます...
makefile `makefile' の読み込み中...
makefile `Util/BuildTools/Vars.mk' の読み込み中 (探索パス) (~ の展開なし)...
makefile `Util/BuildTools/Windows.mk' の読み込み中 (探索パス) (~ の展開なし)...
makefile の更新中....
 ターゲットファイル `Util/BuildTools/Windows.mk' の検討を行います.
  `Util/BuildTools/Windows.mk' のための暗黙ルールを探します.
  語幹 `Windows.mk' とのパターンルールを試します.
  暗黙の必要条件 `Util/BuildTools/Windows.mk.o' を試します.
  語幹 `Windows.mk' とのパターンルールを試します.
  暗黙の必要条件 `Util/BuildTools/Windows.mk.c' を試します.
  語幹 `Windows.mk' とのパターンルールを試します.
  暗黙の必要条件 `Util/BuildTools/Windows.mk.cc' を試します.
  語幹 `Windows.mk' とのパターンルールを試します.
  暗黙の必要条件 `Util/BuildTools/Windows.mk.cpp' を試します.
  語幹 `Windows.mk' とのパターンルールを試します.
  暗黙の必要条件 `Util/BuildTools/Windows.mk.p' を試します.
  語幹 `Windows.mk' とのパターンルールを試します.
  暗黙の必要条件 `Util/BuildTools/Windows.mk.f' を試します.
  語幹 `Windows.mk' とのパターンルールを試します.
  暗黙の必要条件 `Util/BuildTools/Windows.mk.r' を試します.
  語幹 `Windows.mk' とのパターンルールを試します.
  暗黙の必要条件 `Util/BuildTools/Windows.mk.s' を試します.
  語幹 `Windows.mk' とのパターンルールを試します.
  暗黙の必要条件 `Util/BuildTools/Windows.mk.mod' を試します.
  語幹 `Windows.mk' とのパターンルールを試します.
  暗黙の必要条件 `Util/BuildTools/Windows.mk.sh' を試します.
  語幹 `Windows.mk' とのパターンルールを試します.
  暗黙の必要条件 `Util/BuildTools/Windows.mk,v' を試します.
  語幹 `Windows.mk' とのパターンルールを試します.
  暗黙の必要条件 `Util/BuildTools/RCS/Windows.mk,v' を試します.
  語幹 `Windows.mk' とのパターンルールを試します.
  暗黙の必要条件 `Util/BuildTools/RCS/Windows.mk' を試します.
  語幹 `Windows.mk' とのパターンルールを試します.
  暗黙の必要条件 `Util/BuildTools/s.Windows.mk' を試します.
  語幹 `Windows.mk' とのパターンルールを試します.
  暗黙の必要条件 `Util/BuildTools/SCCS/s.Windows.mk' を試します.
  語幹 `Windows.mk' とのパターンルールを試します.
  暗黙の必要条件 `Util/BuildTools/Windows.mk.o' を試します.
  中間ファイル `Util/BuildTools/Windows.mk.o' のルールを探します.
   暗黙ルールの再帰を回避します.
   語幹 `Windows.mk' とのパターンルールを試します.
   暗黙の必要条件 `Util/BuildTools/Windows.mk.c' を試します.
   語幹 `Windows.mk' とのパターンルールを試します.
   暗黙の必要条件 `Util/BuildTools/Windows.mk.cc' を試します.
   語幹 `Windows.mk' とのパターンルールを試します.
   暗黙の必要条件 `Util/BuildTools/Windows.mk.cpp' を試します.
   語幹 `Windows.mk' とのパターンルールを試します.
   暗黙の必要条件 `Util/BuildTools/Windows.mk.p' を試します.
   語幹 `Windows.mk' とのパターンルールを試します.
   暗黙の必要条件 `Util/BuildTools/Windows.mk.f' を試します.
   語幹 `Windows.mk' とのパターンルールを試します.
   暗黙の必要条件 `Util/BuildTools/Windows.mk.r' を試します.
   語幹 `Windows.mk' とのパターンルールを試します.
   暗黙の必要条件 `Util/BuildTools/Windows.mk.s' を試します.
   語幹 `Windows.mk' とのパターンルールを試します.
[... 1143 lines cut ...]
CMake Error: The source directory "C:/carla/Build/osm2odr-visualstudio/x64" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
Error: could not load cache
指定されたファイルが見つかりません。
指定されたファイルが見つかりません。
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "C:\carla\PythonAPI\carla\dependencies\"
成功した子プロセス 0x02f28988 PID 49438056  を回収します
チェインから子プロセス 0x02f28988 PID 49438056 を削除します.
  ターゲットファイル `osm2odr' の再 make に成功しました.
 ターゲットファイル `pythonAPI' の必要条件を満たしました.
ターゲット `pythonAPI' の再 make が必要です.
CreateProcess(NULL,C:/Program Files/Git/bin/sh.exe -c "\"C:/carla/Util/BuildTools/BuildPythonAPI.bat\" --py3",...)
チェインに子プロセス 0x02f28bf8 (pythonAPI) PID 49436696 を取り込みました.
生存子プロセス 0x02f28bf8 (pythonAPI) PID 49436696
-[BuildPythonAPI]: [Batch params]: --py3
Building Python API for Python 3.
C:\Program Files\python312\python.exe: can't open file 'C:\\carla\\PythonAPI\\carla\\setup.py': [Errno 2] No such file or directory

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "C:\carla\PythonAPI\carla\dist"!
成功した子プロセス 0x02f28bf8 PID 49436696  を回収します
チェインから子プロセス 0x02f28bf8 PID 49436696 を削除します.
ターゲットファイル `pythonAPI' の再 make に成功しました.
```

I would appreciate if anyone could help me with the matter.

Nanakea, 2023-11-24:
Please do tell if anyone need more information or is able to help at all.

CMakey, 2023-11-30:
same question... have u finished it? @Nanakea

Nanakea, 2023-12-14:
> same question... have u finished it? @Nanakea

No, I still can't wrap my head around it at all, so I am using VS2019 and wait for the documentation being update or someone replying.

shahbazsiddique69, 2024-11-23:
have you solved it using VS2019? @Nanakea