carla-simulator/carla #6923: Update.bat stuck in download state

dev3225, 2023-11-17:
Stuck here since few hours ( internet connection is fast enough) Anyone have experience with this issue?
[image]

filipthoen2, 2023-11-17:
I see the same issue (CALRA 0.9.15) -- early yesterday (west coast time) things were working fine.

Also trying to clone the UnrealEngine source code:

git clone https://github.com/CarlaUnreal/UnrealEngine.git

generates the error:

D:\UnrealEngine>git clone --depth 1 -b carla https://github.com/CarlaUnreal/UnrealEngine.git .
Cloning into '.'...
info: please complete authentication in your browser...
remote: Repository not found.
fatal: repository 'https://github.com/CarlaUnreal/UnrealEngine.git/' not found

Anybody experiencing this as well?

Thanks,

Filip

filipthoen2, 2023-11-17:
If you look at the link (http://carla-assets.s3.amazonaws.com/20231108_c5101a5.tar.gz) you can see it is hosted on AWS S3 (Simple Storage Service).

I opened the link in a browser on my laptop, and it is a 20+ GB file -- browser started download & indicated it would take about 1 hr to download.

So the link is alive & downloading -- I just restarted 'updsate.bat' on my Linux dev server; let's see how long it will take,

Filip

dev3225, 2023-11-20:
I had the same issue. it was a problem with the '.' at the end for the clone command. It is not needed I guess.

> I see the same issue (CALRA 0.9.15) -- early yesterday (west coast time) things were working fine.
>
> Also trying to clone the UnrealEngine source code:
>
> git clone [...]

dev3225, 2023-11-20:
The issue got resolved itself! automatically.

> Stuck here since few hours ( internet connection is fast enough) Anyone have experience with this issue? [image]

closed 2023-11-20 by dev3225

DanielSioldea, 2024-06-06:
> The issue got resolved itself! automatically.
> 
> > Stuck here since few hours ( internet connection is fast enough) Anyone have experience with this issue? [image]

Hey, how long did this take to resolve itself? Have the same issue.