carla-simulator/carla #6926: Plugin 'CarlaTools' failed to load because module 'CarlaTools' could not be loaded: make launch issue

saitejeswar1, 2023-11-18:
[image]

Lyanyv, 2023-11-18:
Sorry, but i didn't encounter this issue and have no idea...

closed 2023-11-19 by saitejeswar1

chirudeep001, 2023-12-14:
I got this issue when i removed the Houdini engine plugin and restarted the unreal engine

SUEN18, 2023-12-18:
I have this problem too

javageek68, 2024-02-06:
Has anyone found a solution for this?  Or at least have an idea on how to troubleshoot it?  Does Unreal create logs for plugins it fails to load?

mhvicom, 2024-02-08:
I am currently having this issue too.
OS: Windows 11

OYan60718, 2024-03-30:
Same problem
Windows 11, Carla 0.9.15, VS 2019

teoctopus, 2024-07-10:
Try to check the file libsqlite3.so, I replaced the file downloaded from the official website and it doesn't report this error

teoctopus, 2024-07-10:
> Try to check the file libsqlite3.so, I replaced the file downloaded from the official website and it doesn't report this error

carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/lib/libsqlite3.so