carla-simulator/carla #6927: make osmrender in windows 10. Error 1

saitejeswar1, 2023-11-19:
CARLA version: 0.9.15
Platform/OS: Windows 10
Problem you have experienced:  The system cannot find the file specified.
                                                    make: *** [osmrenderer] Error 1
What you expected to happen:
Steps to reproduce: followed this documentation https://carla.readthedocs.io/en/latest/adv_digital_twin/  
Other information (documentation you consulted, workarounds you tried): When I encountered an error in make PythonAPI I followed this thread #6898 

Could someone help me with this? I want to make use of the digital twin tool to create a scene to simulate AD as a part of my studies.

Here is the full log 

C:\carla>make osmrenderer
Cloning into 'C:\carla\Build\libosmscout-source'...
remote: Enumerating objects: 77948, done.
remote: Counting objects: 100% (4287/4287), done.
remote: Compressing objects: 100% (1568/1568), done.
remote: Total 77948 (delta 2735), reused 4154 (delta 2686), pack-reused 73661
Receiving objects: 100% (77948/77948), 80.57 MiB | 2.42 MiB/s, done.
Resolving deltas: 100% (53415/53415), done.
Updating files: 100% (1763/1763), done.
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19045.
-- The C compiler identification is MSVC 19.29.30153.0
-- The CXX compiler identification is MSVC 19.29.30153.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for dlfcn.h
-- Looking for dlfcn.h - not found
-- Looking for fcntl.h
-- Looking for fcntl.h - found
-- Looking for inttypes.h
-- Looking for inttypes.h - found
-- Looking for memory.h
-- Looking for memory.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stdlib.h
-- Looking for stdlib.h - found
-- Looking for strings.h
-- Looking for strings.h - not found
-- Looking for string.h
-- Looking for string.h - found
-- Looking for sys/stat.h
-- Looking for sys/stat.h - found
-- Looking for sys/time.h
-- Looking for sys/time.h - not found
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for unistd.h
-- Looking for unistd.h - not found
-- Looking for C++ include codecvt
-- Looking for C++ include codecvt - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of int16_t
-- Check size of int16_t - done
-- Check size of int32_t
-- Check size of int32_t - done
-- Check size of int64_t
-- Check size of int64_t - done
-- Check size of int8_t
-- Check size of int8_t - done
[... 552 lines cut ...]
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done (3.9s)
-- Generating done (0.0s)
-- Build files have been written to: C:/carla/Build/osm-world-renderer-visualstudio
Microsoft (R) Build Engine version 16.11.2+f32259642 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  1>Checking Build System
  Building Custom Rule C:/carla/osm-world-renderer/CMakeLists.txt
  OsmRenderer.cpp
  Please define _WIN32_WINNT or _WIN32_WINDOWS appropriately. For example:
  - add -D_WIN32_WINNT=0x0601 to the compiler command line; or
  - add _WIN32_WINNT=0x0601 to your project's Preprocessor Definitions.
  Assuming _WIN32_WINNT=0x0601 (i.e. Windows 7 target).
C:\carla\osm-world-renderer\OsmRenderer\include\MapDrawer.h(10,10): fatal error C1083: Cannot open include file: 'osmsc
out/GeoCoord.h': No such file or directory [C:\carla\Build\osm-world-renderer-visualstudio\OsmRenderer.vcxproj]
  MapDrawer.cpp
C:\carla\osm-world-renderer\OsmRenderer\include\MapDrawer.h(10,10): fatal error C1083: Cannot open include file: 'osmsc
out/GeoCoord.h': No such file or directory [C:\carla\Build\osm-world-renderer-visualstudio\OsmRenderer.vcxproj]
  MapRasterizer.cpp
  Generating Code...
The system cannot find the file specified.
make: *** [osmrenderer] Error 1

joppium, 2023-11-21:
I'm having a similar issue. I'm trying to use the Digital Twin Tool on Windows.
I'm pretty sure that the make osmrenderer command is only available on Linux. On Windows, we must use the URL of the OpenStreetMap. However, I'm unable to open the OSM browser in Windows since I don't have the folder CarlaToolsContent/OnroadMapGenerator in the Content browser (only a Carla folder).

Any insight on this will be appreciated, thanks!

domajstorovic, 2023-11-21:
> I'm in the same boat as you. I'm trying to use the Digital Twin Tool on Windows to procedurally generate a 3D environment based on an OSM map.
> 
> I'm pretty sure that the make osmrenderer command is only available on Linux. On Windows, we must use the URL of the OpenStreetMap directly. However, I'm unable to open the OSM browser in Windows since I don't have the folder CarlaToolsContent/OnroadMapGenerator in the Content browser of the Carla UE4Editor (only a Content/Carla folder).
> 
> Through Carla's Digital Twin Tool documentation, it appears that this functionality is available on both Linux and Windows, or did I misinterpret?
> 
> Any insight on this will be appreciated. Thanks!

I am not using Windows, but have you enabled plugins and checked "Show Plugin Content"? 
[image]

joppium, 2023-11-21:
Thanks @domajstorovic! That was it!

saitejeswar1, 2023-11-21:
Thanks @domajstorovic.
@joppium were you able to proceed further with the scene creation?
I am getting errors when I load osm file. [1]
Somehow I am not able to upload the picture of error as well.
I have tried using the URL method as well and it says Map load failed and immediately UE also crashes.[2]

joppium, 2023-11-21:
@saitejeswar1 same here. I'm getting the following error when using the OSM URL:
Assertion failed: Pair != nullptr 
[File:C:\UnrealEngine\Engine\Source\Runtime\Core\Public\Containers/Map.h] [Line: 621]

saitejeswar1, 2023-11-22:
These are the errors I got.
1. It pops up when I use URL and then crashes
[image]
2. It comes up when I use osm file and click on config
[image]

unmanned-harry, 2023-11-27:
> @saitejeswar1 same here. I'm getting the following error when using the OSM URL: Assertion failed: Pair != nullptr [File:C:\UnrealEngine\Engine\Source\Runtime\Core\Public\Containers/Map.h] [Line: 621]

I have the same identical error on windows build, any ideas?

maumai-Yubeen, 2023-11-28:
I have the same problem in Windows 11 build, it seems like a common error with the new feature digital twin @bernatx @Axel1092 
[image]
Version 0.9.15 in Windows 11, 
Assertion failed: Pair != nullptr
[File:C:\UnrealEngine\Engine\Source\Runtime\Core\Public\Containers/Map.h] [Line: 621]

josyulavt, 2024-01-10:
> I filled in the URL (https://www.openstreetmap.org/export#map=16/37.3949/-122.0241) & put '**filip.map**' in the 'File Name:' field:
> 
> This will cause the following error:
> 
> Map load failed. The filename '/Game/CustomMaps/filip.map/filip.map' is not within the game or engine content folders found in 'D:/UnrealEngine/'.
> 
> However, I do see a 'filip' folder being written to:
> 
> <carla_0.9.15 source folder>\carla\Unreal\CarlaUE4\Content\CustomMaps
> 
> The folder is empty however.
> 
> Browsing through '<carla_0.9.15>\carla\Unreal\CarlaUE4\Plugins\CarlaTools\Source\CarlaTools\Private\OpenDriveToMap.cpp' it seems like you can't specify a file extension for the 'File Name' field.
> 
> I tried just 'filip' for the field (no file extension) and now the plug-in starts to render the map. It crashes with the error below but it does generate the following output tree:
> 
> <carla_0.9.15 source folder>\carla\Unreal\CarlaUE4\Content\CustomMaps |-> Filip |-> OpenDrive |- filip.osm |- filip.umap |- filip_Tile_0_0.umap
> 
> It does seem to download the .osm file from the URL I specified (though it seems pretty short).
> 
> Hope this helps -- will try some other options,
> 
> Filip
> ## Crash Report:
> 
> KERNELBASE VCRUNTIME140 UE4Editor_CarlaTools!NILoader::load() UE4Editor_CarlaTools!osm2odr::ConvertOSMToOpenDRIVE() UE4Editor_CarlaTools!UCustomFileDownloader::ConvertOSMInOpenDrive() [D:\users\filip\carla_0.9.15\carla\Unreal\CarlaUE4\Plugins\CarlaTools\Source\CarlaTools\Private\Online\CustomFileDownloader.cpp:41] UE4Editor_CarlaTools!UOpenDriveToMap::ConvertOSMInOpenDrive() [D:\users\filip\carla_0.9.15\carla\Unreal\CarlaUE4\Plugins\CarlaTools\Source\CarlaTools\Private\OpenDriveToMap.cpp:142] UE4Editor_CarlaTools!TBaseUObjectMethodDelegateInstance<0,UOpenDriveToMap,void __cdecl(void),FDefaultDelegateUserPolicy>::ExecuteIfSafe() [D:\UnrealEngine\Engine\Source\Runtime\Core\Public\Delegates\DelegateInstancesImpl.h:611] UE4Editor_CarlaTools!FHttpDownloader::RequestComplete() [D:\users\filip\carla_0.9.15\carla\Unreal\CarlaUE4\Plugins\CarlaTools\Source\CarlaTools\Private\Online\CustomFileDownloader.cpp:127] UE4Editor_CarlaTools!TBaseRawMethodDelegateInstance<0,FHttpDownloader,void __cdecl(TSharedPtr<IHttpRequest,1>,TSharedPtr<IHttpResponse,1>,bool),FDefaultDelegateUserPolicy>::ExecuteIfSafe() [D:\UnrealEngine\Engine\Source\Runtime\Core\Public\Delegates\DelegateInstancesImpl.h:467] UE4Editor_HTTP!TDelegate<void __cdecl(TSharedPtr<IHttpRequest,1>,TSharedPtr<IHttpResponse,1>,bool),FDefaultDelegateUserPolicy>::ExecuteIfBound<void,0>() [D:\UnrealEngine\Engine\Source\Runtime\Core\Public\Delegates\DelegateSignatureImpl.inl:599] UE4Editor_HTTP!FCurlHttpRequest::FinishedRequest() [D:\UnrealEngine\Engine\Source\Runtime\Online\HTTP\Private\Curl\CurlHttp.cpp:1274] UE4Editor_HTTP!FHttpManager::Tick() [D:\UnrealEngine\Engine\Source\Runtime\Online\HTTP\Private\HttpManager.cpp:255] UE4Editor_Core!FTicker::Tick() [D:\UnrealEngine\Engine\Source\Runtime\Core\Private\Containers\Ticker.cpp:95] UE4Editor_Core!<lambda_705c55d452b94d6feba062718920c96b>::operator()() [D:\UnrealEngine\Engine\Source\Runtime\Core\Private\Containers\BackgroundableTicker.cpp:24] UE4Editor_Core!TBaseFunctorDelegateInstance<bool __cdecl(float),FDefaultDelegateUserPolicy,TFunction<bool __cdecl(float)> >::Execute() [D:\UnrealEngine\Engine\Source\Runtime\Core\Public\Delegates\DelegateInstancesImpl.h:831] UE4Editor_Core!FTicker::Tick() [D:\UnrealEngine\Engine\Source\Runtime\Core\Private\Containers\Ticker.cpp:95] UE4Editor!FEngineLoop::Tick() [D:\UnrealEngine\Engine\Source\Runtime\Launch\Private\LaunchEngineLoop.cpp:5045] UE4Editor!GuardedMain() [D:\UnrealEngine\Engine\Source\Runtime\Launch\Private\Launch.cpp:169] UE4Editor!GuardedMainWrapper() [D:\UnrealEngine\Engine\Source\Runtime\Launch\Private\Windows\LaunchWindows.cpp:137] UE4Editor!WinMain() [D:\UnrealEngine\Engine\Source\Runtime\Launch\Private\Windows\LaunchWindows.cpp:268] UE4Editor!__scrt_common_main_seh() [D:\a_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288] kernel32 ntdll

Did you resolve this error ? 

This is what I get 

``` LoginId:6fd06e197a8c41f5b9215eb5d90bdb89-000003e8
EpicAccountId:

Caught signal 6 Aborted

libc.so.6!gsignal(+0xcb)
libc.so.6!abort(+0x12a)
libUE4Editor-InputCore.so!abort_message()
libUE4Editor-InputCore.so!demangling_terminate_handler()()
libUE4Editor-InputCore.so!std::__terminate(void (*)())()
libUE4Editor-InputCore.so!__cxxabiv1::failed_throw(__cxxabiv1::__cxa_exception*)()
libUE4Editor-InputCore.so!__cxa_throw(+0x6e)
libUE4Editor-CarlaTools.so!NILoader::load(OptionsCont&)(+0xc78)
libUE4Editor-CarlaTools.so!osm2odr::ConvertOSMToOpenDRIVE(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, osm2odr::OSM2ODRSettings)(+0x1ecc)
libUE4Editor-CarlaTools.so!UCustomFileDownloader::ConvertOSMInOpenDrive(FString, float, float) [/home/jkrishna/carla/Unreal/CarlaUE4/Plugins/CarlaTools/Source/CarlaTools/Private/Online/CustomFileDownloader.cpp:41]
libUE4Editor-CarlaTools.so!UOpenDriveToMap::ConvertOSMInOpenDrive() [/home/jkrishna/carla/Unreal/CarlaUE4/Plugins/CarlaTools/Source/CarlaTools/Private/OpenDriveToMap.cpp:141]
libUE4Editor-CarlaTools.so!TBaseUObjectMethodDelegateInstance<false, UOpenDriveToMap, void (), FDefaultDelegateUserPolicy>::ExecuteIfSafe() const [/home/jkrishna/UnrealEngine_4.26/Engine/Source/Runtime/Core/Public/Delegates/DelegateInstancesImpl.h:609]
[... cut ...]
UE4Editor!FEngineLoop::Tick() [/home/jkrishna/UnrealEngine_4.26/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:5044]
UE4Editor!GuardedMain(char16_t const*) [/home/jkrishna/UnrealEngine_4.26/Engine/Source/Runtime/Launch/Private/Launch.cpp:171]
libUE4Editor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)()) [/home/jkrishna/UnrealEngine_4.26/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:264]
libc.so.6!__libc_start_main(+0xf2)
UE4Editor!_start()
```

juanrodriguezdekra, 2024-02-07:
Hello! I am encountering the same error when using the digital twin tool. Have you managed to find any type of solution?

josyulavt, 2024-02-07:
> Hello! I am encountering the same error when using the digital twin tool. Have you managed to find any type of solution?

Some people have succeeded to do this by eliminating the buildings from the map more information here https://github.com/carla-simulator/carla/issues/6913

muety, 2024-02-29:
> @saitejeswar1 same here. I'm getting the following error when using the OSM URL: Assertion failed: Pair != nullptr [File:C:\UnrealEngine\Engine\Source\Runtime\Core\Public\Containers/Map.h] [Line: 621]

Same issue here. Any ideas how to work around that?

Blyron, 2024-05-17:
We have fixed this in latest dev branch, please try it

samynjonas, 2024-05-31:
When I try run the tool with a link it seems to crash for me, the error message I receive is the following:
Assertion failed!

Program:
...CarlaTools\Binaries\Win64\UE4Editor-CarlaTools.dll
File:
...\carla...\Geomtry.cpp Line: 37

Expression: _length > 0.0

The link I tried to use is the following:
https://overpass-api.de/api/map?bbox=3.7074,51.0355,3.7185,51.0413

Blyron, 2024-05-31:
It seems like our translator of osm2odr have not created a valid road in that section... 
We need to improve that but we are not gonna be able to fix this in short term. We need to improve it
I apologize

nikolatopic131, 2024-10-15:
> I have the same problem in Windows 11 build, it seems like a common error with the new feature digital twin @bernatx @Axel1092 [...]

I'm having the exact same problem. Did you manage to fix it somehow?