carla-simulator/carla #6932: Traffic lights stay red for long time

CHNtentes, 2023-11-21:
CARLA version: 0.9.13
Platform/OS: Ubuntu 20.04
Problem you have experienced: At crossings, only 1 of 4 lights is green at any time, which is not true in reality. As a result, it takes longt time waiting for green light.
What you expected to happen: Opposite lights should turn green/red together so red time will be shorter.
Steps to reproduce: Just run carlaUE4.sh, then observe traffic lights' behaviour.
Other information (documentation you consulted, workarounds you tried): [This answer](https://github.com/carla-simulator/carla/issues/6121#issuecomment-1387123555) said traffic lights work in group of 2, which is sadly not the case in my simulation.

CHNtentes, 2023-11-24:
1

DnDare, 2024-05-10:
Hi, 
In Carla TrafficLights works as you mention, only one green light at the time.
From the editor you can change the timing for the lights. 
But if you want them to work by pairs the TrafficManager would get you a traffic block, cars would crash each other.

TrafficLights work in groups, with only one active green light at the time. They don't work by pairs. in which 2 TrafficLights have green light at the same time.

closed 2024-05-17 by Madecu