carla-simulator/carla #6935: About Carla in Docker without a screen

ZhuShaoQiang, 2023-11-21:
Hi!
First, here is my env:
Debian 11 server edition(No desktop env),
NVIDIA 1080Ti
Drivier Version 535
Docker Engine - Community  Version:           24.0.7
CARLA version(docker images): carlasim/carla   0.9.13    6fda542451a8   2 years ago    16.6GB

Here is my problems:
First, I run this command follow the doc of Official:
`sudo docker run --privileged --gpus all --net=host -v /tmp/.X11-unix:/tmp/.X11-unix:rw carlasim/carla:0.9.13 /bin/bash ./CarlaUE4.sh -RenderOffScreen`
and the output is :
`4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found`
Then, I try to exec a docker container, and install xdg-user-dirs in this docker container, and In this container, I run this command:
`./CarlaUE4.sh -prefernvidia -RenderOffScreen`,
it outputs:
`4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps`
and there is no process using my GPU。
I have tried to search it all the time, and find nothing
Could you pls help me with this problem? Thans very much!

Mariusmarten, 2024-08-26:
Insane, I was looking into that problem for 2 hours. Simply running: ./CarlaUE4.sh **-nullrhi** -prefernvidia -RenderOffScreen did the trick, thanks a lot!

ZhuShaoQiang, 2024-10-21:
> 我遇到了和你一样的问题，我认为是GPU 容器配置的问题；我是在其他ubunt容器中直接下载运行[Releases]版本； 但是使用-nullrhi 参数，屏蔽GPU使用后就可以了：./CarlaUE4.sh -nullrhi -prefernvidia -RenderOffScreen 。所以应该是启动时的GPU检查不正确导致的，可能与GPU版本有关。

谢谢，我给他装了个虚拟桌面。。。这样无论在命令行中运行还是在哪运行，都会在虚拟桌面里面显示。。很方便。