carla-simulator/carla #6941: Found solution. It occurred by missing libc++-10.  Referring to the developer's intent(In the commit message (branch dev: f66fd8e) that changing to use Unreal's clang.) I didn't change anything in Util/BuildTools/Setup.sh, Just installed libc++ specific version by below code:

ABCSchuetze, 2023-11-22:
Found solution. It occurred by missing libc++-10.  Referring to the developer's intent(In the commit message (branch dev: f66fd8e) that changing to use Unreal's clang.) I didn't change anything in Util/BuildTools/Setup.sh, Just installed libc++ specific version by below code:

```sh
sudo apt update
sudo apt install g++-10
```

Ubuntu 22.04 couldn't run clang-10, but it can install libc++-10 library. So I installed minimum clang version in Ubuntu 22.04(clang-12), and install additionally it.

_Originally posted by @diaa-woo in https://github.com/carla-simulator/carla/issues/6901#issuecomment-1811642752_

diaa-woo, 2023-11-23:
Why did you make this issue? Just curious.

PabloVD, 2025-05-19:
See #6901 for other workarounds.

closed 2025-05-19 by PabloVD