carla-simulator/carla #6942: friction trigger seems doesn't work when we set exact driving command or speed

AOOOOOA, 2023-11-23:
CARLA version: 0.9.13

Platform/OS: Ubuntu 18.03

Problem you have experienced: 

I applied the friction trigger box as the example code in the Carla documentation. I set the friction trigger box as following the ego vehicle just like the spectator. I observed that when I set the ego vehicle as a basic agent or behavior agent and applied control using the "run_step()", which means the ego vehicle is running in autonomous mode. The driving behavior will change with the friction.

However, when I give certain driving commands like giving the exact value of throttle/brake.. or setting the exact driving direction and speed, the driving behavior will not change even with the friction trigger box. Is that normal? @bernatx @nsubiron @marcgpuig @DSantosO 

What you expected to happen:
The friction trigger box will introduce some uncertainties.

Youngzipper, 2024-03-29:
have you solve this problem? i want to try friction trigger to change the behavior of vehicles, too.

Youngzipper, 2024-03-29:
same OS and carla version with you

AOOOOOA, 2024-03-31:
> have you solve this problem? i want to try friction trigger to change the behavior of vehicles, too.
No, it is not important to my task so I just let it go. But I tried several times and the result seems still.

Youngzipper, 2024-03-31:
> > have you solve this problem? i want to try friction trigger to change the behavior of vehicles, too.
> > No, it is not important to my task so I just let it go. But I tried several times and the result seems still.

ok thank you