carla-simulator/carla #6943: Carla 0.9.15: Cannot create rpc server for Traffic Manager

haabe, 2023-11-23:
CARLA version: 0.9.15
Platform/OS: Windows

Problem you have experienced: 
When I run Python example scripts, such as `performance_benchmark.py`, I get the following runtime error: `RuntimeError: trying to create rpc server for traffic manager; but the system failed to create because of bind error.`
What you expected to happen: The script to perform without any errors.

Steps to reproduce:
1. Download and build UE from source (Carla branch)
2. Clone and build Carla from source
3. Open CarlaUE4 project and run the simulation from the UE Editor
4. Open a X64 native command prompt for VS 2019, and launch the Python script.

I've tried using both Python 3.7.7 and Python 3.8.18 with the same result. Using Python 3.7.7 and Carla 0.9.13 did not result in this error, however.

haabe, 2023-11-24:
I tried using the precompiled version, and experienced the same thing:

1. Download and extract carla_0.9.15.zip + AdditionalMaps_0.9.15.zip
2. Clear any Carla-related environment variables
3. Generate a new Conda environment for Python 3.7 (´conda create -n py3.7 python=3.7 anaconda`)
4. Activate new environment (`activate py3.7`)
5. Add Carla client to environment (`pip3 install \WindowsNoEditor\PythonAPI\carla\dist\carla-0.9.15-cp37-cp37m-win_amd64`)
6. Execute CarlaUE4.exe
7. Execute a script (`python \WindowsNoEditor\PythonAPI\examples\generate_traffic.py`)

Result:
```
Traceback (most recent call last):
  File "WindowsNoEditor\PythonAPI\examples\generate_traffic.py", line 374, in <module>
    main()
  File "WindowsNoEditor\PythonAPI\examples\generate_traffic.py", line 167, in main
    traffic_manager = client.get_trafficmanager(args.tm_port)
RuntimeError: trying to create rpc server for traffic manager; but the system failed to create because of bind error.
```

haabe, 2023-11-29:
So investigating further, I found that Carla uses port `:8000` as default for RPC. This is hidden away in the documentation, under [Running multiple Traffic Managers](https://carla.readthedocs.io/en/latest/adv_traffic_manager/#traffic-manager-servers-and-clients).

This led me to identify which process occupying port 8000, which for me was IncrediBuild. Resolving this conflict made Carla successfully create the rpc server.

Now I wish the documentation would be easier to scan through for such default settings.

closed 2023-11-29 by haabe

haabe, 2024-01-15:
> Hi,
> 
> I have the same issue and I cannot find any process using 8000. My carla runs on port 2000. Your help would be appreciated. OS: windows 10 carla: 9.15 script: manual_control.py

In my case it was part of the IncrediBuild system for Visual Studio that blocked the port. You could try a tool like [TCPView for Windows](https://learn.microsoft.com/en-us/sysinternals/downloads/tcpview) to see if you can identify which app is blocking the port.

ssingh17j, 2024-08-23:
the rpc uses port 8000 by default.
I got a similar issue , as port 8000 was occupied, I killed the task on that particular port using the PID and it worked for me.