carla-simulator/carla #6946: CARLA with 2 fps on linux and 60 on windows on the same machine

DiogoValdrez, 2023-11-23:
CARLA version: 0.9.15
Platform/OS: Linux (Ubuntu22)

Problem you have experienced: 

When I run packaged version in linux I get 2 fps.
I have the drivers of nvidia installed (nvididia-driver-535) and my pc is not that bad (nvme, intelcore i7 10th gen, nvidia gtx1660ti).
In windows, in the same machine, I can get easily 60 fps so the problem is not the hardware.
I have already tried the same thing on version 0.9.13 but I have the same problem.

Steps to reproduce:
Download the packaged version of carla
run ./CarlaUE4.sh -quality-level={Low}

oliver-schumann, 2024-02-13:
I have a similar problem with the Carla simulator but with two Linux PCs. One with two 2080 TI and one with a 3090. Both have AMD Thread rippers inside. So I would say hardware is no Problem.

The PC with the 2080 runs smoothly with approx. 20 fps. The other one with 1-2 fps. This happens also in multi GPU mode in the first one with two 2080s. Something seems to be wrong here...

Blyron, 2024-04-19:
Are you sure that you are using your dedicated GPUs when running carla in linux? 
You can set graphics card used by Graphics adapter with -graphicsadapter=1/2/3 , 0 may be indentified as integrated graphics adapter

Blyron, 2024-04-26:
Hello again, are you happy to close this issue?

oliver-schumann, 2024-04-26:
Yes I'm sure. In nvtop I see either one card running normally or both cards running in multi GPU mode. In both cases with approx 40%.

To narrow the bug downthis happens only if LIDAR sensors are used, at least in my case.

I would not close an Issue if the error still persists, as it wouldn't make any sense.

Blyron, 2024-04-26:
This may be CPU related then. LIDAR is CPU based and uses tons of CPU. 
Could you give more info about CPU?

oliver-schumann, 2024-04-26:
Okay, I will try to do some tests when I get back in the office next week.

Blyron, 2024-05-30:
As there is no info we are gonna close this issue

closed 2024-05-30 by Blyron