carla-simulator/carla #6953: Influence velocity and acceleration

Aaroniaa, 2023-11-28:
Hi,

I'm currently trying to simulate and test various driving scenarios with Carla. Therefore, I would like to specify certain velocity and acceleration values for the vehicle. So far, I have only been able to influence the behaviour of the car via driving parameters such as throttle, steering or gear shifting. Is it possible to set the car's acceleration and velocity directly?

I've seen that it might be about "apply_ackermann_control()", but couldn't figure out how to use it. 

Many thanks in advance!

muety, 2023-12-18:
Same thing here, the `apply_ackermann_control()` method seemed like a practical way to set a target speed and steering and have the controller steer the car then. However, it doesn't behave as I would expect it. When setting a speed of, e.g., 5 m/s, the car doesn't move at all, only "stutters" a bit. I'm not very versed in control theory, so might well just have misunderstood some of the concepts there.

muety, 2023-12-18:
Either I'm totally wrong about how to use that function or it is buggy. Given the exact same code, it seems work, while sometimes doesn't. Here's a little "demo".

https://github.com/carla-simulator/carla/assets/8978843/2453884f-e09c-45b0-af21-d2d91fd36c44

I'm using CARLA 0.9.14.

Fillipko, 2025-06-12:
I was having a similar issue with inconsistent runs and the vehicle stuttering. To resolve the issues, you'll need a couple of things: 

1. Set a fixed time step for the simulation to get rid of inconsistent runs:
```python
settings = world.get_settings()
settings.fixed_delta_seconds = 0.02
world.apply_settings(settings)
```

2. Manually set the vehicle into forward gear. Turns out the automatic transmission causes some initial stuttering when going from park into forward gear:
```python
control = carla.VehicleControl()
control.manual_gear_shift = True
control.gear = 1
ego.apply_control(control)
```

closed 2025-10-18 by Blyron