carla-simulator/carla #6962: error occur when make launch.  InitSDL() failed, cannot create application instance.

re-frank, 2023-11-30:
CARLA version: 
0.9.15(latest)

Platform/OS: 
docker taged "pytorch/pytorch   1.12.1-cuda11.3-cudnn8-devel" ,and run by “docker run -it --shm-size 64g -v /data/4TSSD/**_docker_workspace:/workspace/ --name **_carla --gpus all -e NVIDIA_DRIVER_CAPABILITIES=all fa50f7fed43a”
and the container is run in the remote server without head by ssh in vscode.
Problem you have experienced: 
When i make launch in carla, the error will occur (just show the part of error)：
[2023.11.29-19.01.05:247][  0]LogInit: Unix-specific commandline switches:
[2023.11.29-19.01.05:247][  0]LogInit:  -ansimalloc - use malloc()/free() from libc (useful for tools like valgrind and electric fence)
[2023.11.29-19.01.05:247][  0]LogInit:  -jemalloc - use jemalloc for all memory allocation
[2023.11.29-19.01.05:247][  0]LogInit:  -binnedmalloc - use binned malloc  for all memory allocation
[2023.11.29-19.01.05:247][  0]LogInit:  -filemapcachesize=NUMBER - set the size for case-sensitive file mapping cache
[2023.11.29-19.01.05:247][  0]LogInit:  -useksm - uses kernel same-page mapping (KSM) for mapped memory (OFF)
[2023.11.29-19.01.05:247][  0]LogInit:  -ksmmergeall - marks all mmap'd memory pages suitable for KSM (OFF)
[2023.11.29-19.01.05:247][  0]LogInit:  -preloadmodulesymbols - Loads the main module symbols file into memory (OFF)
[2023.11.29-19.01.05:247][  0]LogInit:  -sigdfl=SIGNAL - Allows a specific signal to be set to its default handler rather then ignoring the signal
[2023.11.29-19.01.05:247][  0]LogInit:  -httpproxy=ADDRESS:PORT - redirects HTTP requests to a proxy (only supported if compiled with libcurl)
[2023.11.29-19.01.05:247][  0]LogInit:  -reuseconn - allow libcurl to reuse HTTP connections (only matters if compiled with libcurl)
[2023.11.29-19.01.05:247][  0]LogInit:  -virtmemkb=NUMBER - sets process virtual memory (address space) limit (overrides VirtualMemoryLimitInKB value from .ini)
[2023.11.29-19.01.05:247][  0]LogInit: Initializing SDL.
[2023.11.29-19.01.05:249][  0]LogInit: Warning: Could not initialize SDL: No available video device
[2023.11.29-19.01.05:249][  0]LogInit:  - Physical RAM available (not considering process quota): 252 GB (257512 MB, 263692460 KB, 270021079040 bytes)
[2023.11.29-19.01.05:249][  0]LogInit:  - VirtualMemoryAllocator pools will grow at scale 1.4
[2023.11.29-19.01.05:249][  0]LogInit:  - MemoryRangeDecommit() will be a no-op (re-run with -vmapoolevict to change)
[2023.11.29-19.01.05:285][  0]LogInit: Physics initialised using underlying interface: PhysX
[2023.11.29-19.01.05:313][  0]LogInit: Using OS detected language (en-US).
[2023.11.29-19.01.05:313][  0]LogInit: Using OS detected locale (en-US).
[2023.11.29-19.01.05:314][  0]LogTextLocalizationManager: No specific localization for 'en-US' exists, so the 'en' localization will be used.
[2023.11.29-19.01.05:504][  0]LogInit: Initializing SDL.
[2023.11.29-19.01.05:507][  0]LogInit: Warning: Could not initialize SDL: No available video device
[2023.11.29-19.01.05:507][  0]LogInit: Warning: FLinuxSplashState::InitSplashResources() : InitSDL() failed, there will be no splash.
[2023.11.29-19.01.05:507][  0]LogInit: Initializing SDL.
[2023.11.29-19.01.05:510][  0]LogInit: Warning: Could not initialize SDL: No available video device
[2023.11.29-19.01.05:510][  0]LogInit: Error: FLinuxApplication::CreateLinuxApplication() : InitSDL() failed, cannot create application instance.
[2023.11.29-19.01.05:510][  0]LogCore: FUnixPlatformMisc::RequestExit(bForce=true, ReturnCode=1)
[2023.11.29-19.01.05:510][  0]LogCore: FUnixPlatformMisc::RequestExit(1)

What you expected to happen: 
I want to install carla in my container, so that i can experiment ReSimAD. So i need run "make launch" successfully first, then i want to display the video in web browser in my Mac by carlaviz.

Steps to reproduce:
I have built carla following the official course about "Linux build". And when i run "cd ~/UnrealEngine_4.26/Engine/Binaries/Linux && ./UE4Editor " with "-RenderOffScreen", something wrong with UE4.26 either, like :
LogModuleManager: Warning: ModuleManager: Unable to load module 'ExampleDeviceProfileSelector'  - 0 instances of that module name found.
LogInit: Initializing SDL.
LogInit: Hinting SDL to use 'dummy' video driver.
LogInit: Initialized SDL 2.0.12 revision: 13609 (hg-13609:34cc7d3b69d3) (compiled against 2.0.12)
LogInit: Using SDL video driver 'dummy'
LogInit: Display metrics:
LogInit:   PrimaryDisplayWidth: 1024
LogInit:   PrimaryDisplayHeight: 768
LogInit:   PrimaryDisplayWorkAreaRect:
LogInit:     Left=0, Top=0, Right=1024, Bottom=768
LogInit:   VirtualDisplayRect:
LogInit:     Left=0, Top=0, Right=1024, Bottom=768
LogInit:   TitleSafePaddingSize: X=0.000 Y=0.000 Z=0.000 W=0.000
LogInit:   ActionSafePaddingSize: X=0.000 Y=0.000 Z=0.000 W=0.000
LogInit:   Number of monitors: 1
LogInit:     Monitor 0
LogInit:       Name: 0
LogInit:       ID: display0
LogInit:       NativeWidth: 1024
LogInit:       NativeHeight: 768
LogInit:       bIsPrimary: true
LogLinux: Warning: MessageBox: Plugin 'ExampleDeviceProfileSelector' failed to load because module 'ExampleDeviceProfileSelector' could not be found.  Please ensure the plugin is properly installed, otherwise consider disabling the plugin for this project.: Message: That operation is not supported: 
Message dialog closed, result: Cancel, title: Message, text: Plugin 'ExampleDeviceProfileSelector' failed to load because module 'ExampleDeviceProfileSelector' could not be found.  Please ensure the plugin is properly installed, otherwise consider disabling the plugin for this project.
LogCore: Engine exit requested (reason: EngineExit() was called)
LogExit: Preparing to exit.
LogModuleManager: Shutting down and abandoning module ChunkDownloader (72)
LogModuleManager: Shutting down and abandoning module LauncherChunkInstaller (70)
LogModuleManager: Shutting down and abandoning module OnlineBlueprintSupport (68)
LogModuleManager: Shutting down and abandoning module OnlineSubsystem (64)
LogModuleManager: Shutting down and abandoning module HTTP (59)
LogHttp: Display: cleaning up 0 outstanding Http requests.
LogModuleManager: Shutting down and abandoning module SSL (58)
LogModuleManager: Shutting down and abandoning module NiagaraVertexFactories (54)
LogModuleManager: Shutting down and abandoning module NiagaraShader (52)
LogModuleManager: Shutting down and abandoning module ChaosCloth (50)
LogModuleManager: Shutting down and abandoning module VariantManagerContent (48)
LogModuleManager: Shutting down and abandoning module DatasmithContent (46)
LogModuleManager: Shutting down and abandoning module AISupportModule (44)
LogModuleManager: Shutting down and abandoning module PlatformCryptoOpenSSL (42)
LogModuleManager: Shutting down and abandoning module PlatformCryptoTypes (40)
LogModuleManager: Shutting down and abandoning module PlatformCrypto (38)
LogModuleManager: Shutting down and abandoning module AnimationModifiers (36)
LogModuleManager: Shutting down and abandoning module AudioEditor (34)
LogModuleManager: Shutting down and abandoning module PropertyEditor (33)
LogModuleManager: Shutting down and abandoning module TextureCompressor (30)
LogModuleManager: Shutting down and abandoning module RenderCore (28)
LogModuleManager: Shutting down and abandoning module Landscape (26)
LogModuleManager: Shutting down and abandoning module SlateRHIRenderer (24)
LogModuleManager: Shutting down and abandoning module OpenGLDrv (22)
LogModuleManager: Shutting down and abandoning module AnimGraphRuntime (20)
LogModuleManager: Shutting down and abandoning module Renderer (18)
LogModuleManager: Shutting down and abandoning module Engine (16)
LogModuleManager: Shutting down and abandoning module CoreUObject (14)
LogModuleManager: Shutting down and abandoning module NetworkFile (12)
LogModuleManager: Shutting down and abandoning module CookedIterativeFile (10)
LogModuleManager: Shutting down and abandoning module StreamingFile (8)
LogModuleManager: Shutting down and abandoning module SandboxFile (6)
LogModuleManager: Shutting down and abandoning module PakFile (4)
LogModuleManager: Shutting down and abandoning module RSA (3)
LogExit: Exiting.
LogInit: Tearing down SDL.
Exiting abnormally (error code: 1)
I think the point is the platform, but i don’t know how to deal with it.  
Please help me !
Online anytime !
Other information (documentation you consulted, workarounds you tried):

Anderw-S, 2024-04-24:
Hi I've got the same issue. Have you solved it? I'm guessing that its just we're deploying it on a remote server and we don't have a display for UE4 to connect to.

NBRodrigues, 2024-05-04:
Any solution

gawinwong, 2024-06-13:
> Could not initialize SDL: No available video device
Check your nvidia driver and vulkan: apt install libvulkan1