carla-simulator/carla #6963: generate_opendrive_world does not block till environment is ready

fgrzeszc, 2023-11-30:
**CARLA version:** 0.9.13
**Platform/OS:** Ubuntu 20.04
**Problem you have experienced:**  when running scenario using OpenDrive file, scenario execution starts before world is fully initialized in Carla server. 
**What you expected to happen:** `generate_opendrive_world` blocks until world is fully initialized, resulting in scenario starting execution when environment is ready
**Steps to reproduce:** 

1. use OpenDrive map with long road (e.g., 10km).
2. run scenario_runner using that map
3. scenario starts execution and simulation time flows, even though Carla API does not return active agent

It is visible in console logs from Carla that server is stuck for long time on `Create detail mesh...` step.
```
Loading...
Building...
Building navigation:
- 33333 x 20 cells
- 0.0K verts, 0.0K tris
Finding walkable triangles...
Filter walkable surfaces...
Partition of walkable surfaces...
Remove walkable surfaces by agent raius...
Defining areas by material
Trace and simplify region contours...
WARNING: cannot parse georeference: ''. Using default values. 
Build polygons mesh from contours...
Create detail mesh... 
```

**Other information (documentation you consulted, workarounds you tried):**
1. preload OpenDrive map with `python /opt/carla/PythonAPI/util/config.py -x  ACC_Straight_Freeway_CCRM_20.xodr`
2. `sleep 30`
3. execute scenario

josyulavt, 2024-02-01:
Hi, 
It's been a while, did you get to solve this issue? I'm facing the same problem currently. Thank you

tst-777, 2024-02-01:
still using the workaround to preload the world...

josyulavt, 2024-02-01:
Thanks for the response!  I get this error when I use the workaround it got stuck for a long time this way.

```
TileRow 52/129: 16 tiles added (total 416)
TileRow 53/129: 12 tiles added (total 428)
TileRow 54/129: 8 tiles added (total 436)
TileRow 55/129: 6 tiles added (total 442)
TileRow 56/129: 8 tiles added (total 450)
TileRow 57/129: 8 tiles added (total 458)
TileRow 58/129: 7 tiles added (total 465)
TileRow 59/129: 7 tiles added (total 472)
TileRow 60/129: 8 tiles added (total 480)
TileRow 61/129: 7 tiles added (total 487)
TileRow 62/129: 7 tiles added (total 494)
TileRow 63/129: 10 tiles added (total 504)
TileRow 64/129: 12 tiles added (total 516)
```
I had to add georeference tag for this to work. 

and `client.generate_opendrive_world(osm_content)` fails for me with the error 

```
Loading /home/ship/carla/CarlaUE4/Content/Carla/Maps/Nav/OpenDriveMap.obj'...
WARNING: cannot parse georeference: ''. Using default values. 
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=196744 
Malloc Size=131160 LargeMemoryPoolOffset=327928 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
```
any ideas on how to fix this? how did you add a georeference to osm map? thank you