carla-simulator/carla #6976: The following error occurs when I make PythonAPI. CMake could not find ZLIB. Is there any solution?

Narcissus1004, 2023-12-06:


Narcissus1004, 2023-12-06:
Platform:
--     Host: Windows-10.0.14393 AMD64
--     Target: Windows-10.0.14393 AMD64
--     CMake: 3.28.0-rc5
--     CMake generator: Visual Studio 16 2019
--     CMake build tool: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/MSBuild/Current/Bin/MSBuild.exe
--     Compiler: MSVC 19.29.30153.0
--
-- Found Proj: C:/Users/carla/Desktop/carla-0.9.12/Build/proj-install/lib/proj.lib
-- Could NOT find ZLIB (missing: ZLIB_LIBRARY ZLIB_INCLUDE_DIR)
-- Enabled features: Windows-10.0.14393 AMD64 MSVC 19.29.30153.0 Release Proj
-- Configuring done (0.3s)
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
ZLIB_LIBRARY
    linked by target "osm2odr" in directory C:/Users/carla/Desktop/carla-0.9.12/Build/om2odr-source/src

-- Generating done (0.2s)
CMake Generate step failed.  Build files cannot be regenerated correctly.

andrejfsantos4, 2023-12-14:
A temporary workaround that seems to have fixed the issue for me is to add a new file called `FindZLIB.cmake` inside `carla\Build\om2odr-source\build\cmake_modules` with this content:

```
set(ZLIB_INCLUDE_DIRS "<path-to-carla>/Build/zlib-install/include")
set(ZLIB_FOUND True)
set(ZLIB_VERSION 1.2.13)  # confirm that this is your version 
```

Alternatively, skip the client build process (which usually throws multiple errors in my experience) and install the CARLA client Python package with `pip install carla`. For many applications you'll only need to build the CARLA server.

hcd-experiment, 2024-02-13:
I tried this but it still gave an error. Was able to solve it by removing the `S` from `ZLIB_INCLUDE_DIRS`