carla-simulator/carla #6979: Digital twin map generation crash

AbdelrhmanRady, 2023-12-06:
My team wants to build a map for our campus(which is not realtively big at all)  using the new digital twin application for a graduation project, we've been trying for weeks to get it running, on linux/windows with different computers (with good specs), the output is always the same, the unreal engine crashes at the end with no usable output

We followed the documentation(https://carla.readthedocs.io/en/latest/adv_digital_twin/#digital-twin-tool)  carefully and thoroughly, so If there's someone who managed to get it running we would appreciate any tips or tricks

dylanyd, 2023-12-07:
I also got the unreal engine crash when I tried to generate the map from OpenStreetMap URL, I got the following error:
[image]

Blyron, 2024-05-17:
It should be fixed now with latest dev version.

closed 2024-05-17 by Blyron