carla-simulator/carla #6982: Packaging a Custom Map in CARLA Build from Source on Linux for Binary Version of CARLA on Windows

DingZhiKun66, 2023-12-07:
CARLA version: 0.9.14
Platform/OS: ubuntu 18.04 for build version and windows 10 for binary version

I want to package a custom map I designed in build version of CARLA and use it in binary versions. My build version of CARLA is on ubuntu 18.04 and I successfully exported the package and used it in binary version of CARLA on Linux, using instructions [here](https://carla.readthedocs.io/en/latest/tuto_A_create_standalone/). However, the exported package does not work on binary version of CARLA on windows. 

I copied the assets and the map folder from the compressed package, the map load correctly,but when I spawned vehicles cannot stand on the surface and they go down.

csonthomihaly, 2024-02-23:
Did it work inside the built version of Carla?
In my case sometimes the problem was, that the spawning point was below the road level.

I do not know how does it work on Windows, but in Ubuntu ImportAssets script works fine for this task.

Blyron, 2025-08-25:
I do not think it is possible because cooking code is done for each platform so if you want to cook for windows version you need windows built from source